andysworkshop/stm32plus #82: use system include files

punkkeks, 2015-01-11:
The question is why you use your own stl files in '/usr/local/arm-none-eabi/include/stm32plus-030600/stl'? Why do you don't use the system side include files?
That is a problem, because I want to use std::bind and that's not in your provided file. Is it really necessary? I hope not

andysworkshop, 2015-01-11:
Hi, the STL included with stm32plus is based on the original, simple, SGI implementation with customisations to make it suitable for use on a resource-limited MCU. The standard library, libstdc++, included with gcc is designed for devices that have resources comparable to a PC and is not suitable for small MCUs.

`std::bind` wasn't in the SGI implementation (I think it was brought in for C++0x from boost). If you want to extract `std::bind` from the `functional` header in libstdc++ and add it to `functional` in the stm32plus headers then I'd be OK with that.

mikepurvis, 2015-01-12:
Note that if you don't need to actually bind data and just want pointers to member functions, you can use FastDelegate: https://github.com/andysworkshop/stm32plus/tree/master/lib/include/event/fd

All of the event notification callbacks in stm32plus accept a FastDelegate bind, for example: https://github.com/andysworkshop/stm32plus/blob/e172e1d0d7d4d343500115db0e3ee69270ded2bb/examples/net_dns/net_dns.cpp#L123

punkkeks, 2015-01-14:
Okay, there is a small problem. Sometimes I want to use some c++11 classes/functions like bind(okay, I see you have a alternative), std::array (you could use old plane c array, but I do not want to) or the most important one for me the random generators. 
Does somebody ever measured the overhead by using the sgi vs the libstdc++ imeplementation? Maybe the clang project libc++ could be an alternative? (I tried it once, worked like a charme)
So, at least I'd like to have an option to say, let me have the stdlib, nevermind what you (compiler) think :) Does it sound like a plan?

andysworkshop, 2015-01-14:
You can change your include path to replace stm32plus/include/stl with the path that points at your preferred alternative. Compatibility issues will be your problem to solve though.

closed 2015-01-14 by andysworkshop