andysworkshop/stm32plus #84: Unwanted change in Config/stdperiph.h ?

joaoc, 2015-01-14:
Hi, while navigating through the source code i notice a change in this file that i can't find in the file history and that doesn't make sense for me.

In the Config/stdperiph.h file the line 77 used to be
```c++ 
#elif defined(STM32PLUS_F0_50)
```
but now its just
```c++
#elif defined(STM32PLUS_F0)
```

Is this intended?

andysworkshop, 2015-01-15:
I think this might have happened during the merge of the `usb` branch into `master`. The merge was clean and completely automatically but for some reason it must have favoured what was in the `usb` branch. 

Certainly happy to put it back. I don't think that it could have happened in many other places because outside of the high-level config includes the `usb` changes were restricted to new folders and the f4 standard peripheral library.

PS. Did you mean `#elif defined(STM32PLUS_F0_51)` ?

joaoc, 2015-01-15:
yes, that 50 was a typo...

commit 5d5a4a1c (2015-01-16): fix F0 include block. closes #84
closed 2015-01-16 by andysworkshop