andysworkshop/stm32plus #87: USB peripheral on non-8MHz crystals

mikepurvis, 2015-01-22:
I tried the usb_device_cdc_com_port example on my STM32F407 board, and it seems to halt during the `usb.initialise(usbParams)` call. Still investigating further, but my board is basically the reference design apart from using a 25MHz crystal.

Will investigate and report further.

andysworkshop, 2015-01-22:
I just did a quick re-check of the CDC example on the discovery board and all is fine on Windows 8.1. Your development system is Linux isn't it? It shouldn't make a difference of course but the CDC example on Windows does require a driver INF file and I don't know what the Linux equivalent would be.

Have you tried the other examples to narrow the problem down to the CDC device class? e.g. the `usb_device_hid_keyboard` example only needs an external button to work and the `usb_device_msc_internal` example is completely standalone. Neither of those require additional drivers.

mikepurvis, 2015-01-22:
Hmm. Unfortunately, all my disco boards have been switched to 25MHz crystals now, but I can get another one to compare.

Definitely no configuration needed on the Linux side— I have been previously using this exact hardware setup with ST's somewhat broken and unstable USB code.

mikepurvis, 2015-01-23:
The plot thickens. On an unmodified 407 Discovery board:

 - I can load the usb_device_cdc_com_port example, and it properly enumerates when plugged in.
 - If I try to use a terminal program to talk to it, the 407 crashes and will not enumerate again until reset.
 - If I unplug and replug the USB port from it, it crashes my Ubuntu Trusty VM.

mikepurvis, 2015-01-23:
Okay, some more details have been edited into the post above. Note that this is all with the 8MHz crystal board—none of these examples work at all on the 25MHz crystal, so I think there are still some timing assumptions built into the library.

andysworkshop, 2015-01-23:
Interesting... crashing the VM is particularly impressive. I've got Ubuntu and OS X in a VM here (VMWare Player) and will give it a go this weekend. Did you try any of the other examples that demo the HID and MSC device classes? It would be interesting to know if the issues are limited to one particular device class.

mikepurvis, 2015-01-23:
I tried the msc_internal one, with similar results.

Just gave the keyboard example a try. AFAICT, it is fully functional in OS X and Ubuntu, though only with the 8MHz board— the 25MHz board never enumerates.

mikepurvis, 2015-01-23:
(ping for updated post above)

andysworkshop, 2015-01-24:
I was able to reproduce the issue you illustrated with RealTerm and have fixed that which also seems to have fixed the CDC device on Linux.

Testing on Ubuntu Server 64 in VMWare Player:

	andy@marge:~$ lsusb
	Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
	Bus 002 Device 006: ID dead:0010  
	Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
	Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
	Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

	andy@marge:~$ ls -l /dev/ttyA*
	crw-rw---- 1 root dialout 166, 0 Jan 24 09:38 /dev/ttyACM0

	andy@marge:~$ sudo screen /dev/ttyACM0 9600

        [ goes into full screen mode echoing back chars as expected ]

I don't have news on the 25MHz issue yet.

andysworkshop, 2015-01-24:
Can you check that your startup settings result in a 48MHz USB clock?

    25000000 * (PLL_N / PLL_M ) / PLL_Q = 48000000

mikepurvis, 2015-01-26:
Thanks for the fix. I tried a different 25MHz board, which now works great, so I think there's some issue more specific to my hardware config. Will raise a new ticket if necessary.

closed 2015-01-26 by mikepurvis