andysworkshop/stm32plus #90: Trouble with USB/CDC

mikepurvis, 2015-01-29:
This is my custom board again, a continuation of #87.

I'm now making it through the initialization function, and I get the following [eventType](https://github.com/andysworkshop/stm32plus/blob/f2e5ea7be9072513f04b45c250473545eb311fe9/lib/include/usb/UsbEventDescriptor.h#L18)s. sent to my `onEvent` handler:

```
DEVICE_INIT (0)
DEVICE_START (2)
OTG_FS_INTERRUPT (25)
DEVICE_IRQ_RESET (17)
DEVICE_OPEN_ENDPOINT (4)
DEVICE_OPEN_ENDPOINT (4)
CLASS_DEINIT (27)
DEVICE_CLOSE_ENDPOINT (5)
DEVICE_CLOSE_ENDPOINT (5)
DEVICE_CLOSE_ENDPOINT (5)
```

It's exactly this set of events whether the port is connected to a host or not, and no further events are emitted by disconnecting/connecting the port after startup.

I believe the hardware is correct, as this board works great with the ROM bootloader, and also with the ST library's USB code.

Any thoughts what could be going on here?

andysworkshop, 2015-01-31:
That sequence of events is correct in that it's exactly what I get. The two endpoints being opened are the IN and OUT directions for the bi-directional control endpoint 0. The last part of the `USB_LL_Reset`  function then raises a `CLASS_DEINIT` event which de-initialises our 3 CDC endpoints. Of course they haven't actually been initialised yet but that doesn't do any harm, it's just the way ST's flow of control works.

The next thing that happens should be that lots of interrupts should start being raised when the device is plugged in and clearly this is not happening for you. I'm still looking into how that can happen and am wondering if it's somehow in sleep mode.

mikepurvis, 2015-02-08:
Hmm, I'm clearly getting at least those initial interrupts, and I'm basically initializing this peripheral last after anything else. But I suppose a reasonable next step would be dropping other peripherals until this one starts working.

It's also possible that this is RTOS-related; I can try disabling the RTOS startup and seeing if the peripheral behaves when the main program is just a single thread spinning.

andysworkshop, 2015-02-08:
I've just modded one of my discovery boards with a 25MHz crystal & load caps in place of the standard 8MHz item so I'll get a chance to try to replicate this issue over the next few days.

mikepurvis, 2015-02-09:
Oh, awesome.

Will be curious what you discover— I'm not able to get the 25MHz CDC demo working on either the disco board or my own, though my notes show that the 25MHz MSC demo definitely was working on at least the disco board at some point, and I don't seem to have that working at the moment either.

It was inconclusive, but I also noted previously that some of this seemed to depend how exactly the firmware was flashed— openocd (stlinkv2), st-util (stlinkv2), and dfu-util (ROM bootloader) all seem to do not quite the same thing as each other.

mikepurvis, 2015-02-09:
Oh, and to update the above remarks— no amount of disabling other peripherals or the RTOS seemed to make a difference. And I don't believe it's an issue with my PC, as re-flashing the firmware based on ST's libraries restores full functionality.

andysworkshop, 2015-02-10:
I just tested the `usb_device_hid_mouse` example on my modded 25MHz discovery board and it works  fine on Windows, no issues at all. I don't have much time now but will try the CDC example tomorrow to see if that's got any unique issues. I also have another 3rd party F407 board with a 25MHz crystal that I can try it on.

The modded discovery board has the crystal and load caps replaced as well as R24 replaced with 0R and R68 removed.

andysworkshop, 2015-02-11:
I've now tested `usb_device_hid_mouse`, `usb_device_hid_keyboard` and `usb_device_cdc_com_port` on my 25MHz modded discovery board and also on the 'PowerMCU' 25MHz F407 board (seen in [this article](http://andybrown.me.uk/wk/2014/09/09/stlinkv2/)). All worked perfectly on Windows. I also tested the CDC example on Linux in a VM and that worked just as well. There was one minor (1 byte) buffer overrun issue in the CDC example that I've fixed and pushed but this would not interfere with the device enumeration.

The boards were flashed with OpenOCD 0.8.0 using the `-f scripts/board/stm32f4discovery.cfg` option.

I think we need to look a little more closely at your board because it's clearly not the 25MHz crystal that's the issue. What kind of board is it exactly? Are there any onboard peripherals that could interfere with PA9..12?

We can try to see if your PHY has gone into low power mode. Change the PHY declaration from `InternalFsPhy<>` to `InternalFsPhy<FsLowPowerFeature>` and ensure that the `SetSysClock()` C function in the example's `System.c` file is non-static. When you plug a device into the host you should see `onWakeupEvent()` in the `FsLowPowerFeature` class get called. Can you see if that's happening?

mikepurvis, 2015-02-11:
This is really bizarre. I think I had a (partially?) malfunctioning discovery kit which caused some red herrings here, for which I am very sorry— with a different 25MHz-equipped one, the `usb_device_cdc_com_port` example works great.

*However*:

 - The `usb_device_cdc_com_port` example doesn't work on my custom board, and
 - The custom firmware which integrates with `usb_device_cdc_com_port` doesn't work on the disco kit.

I'm currently in the process of stripping my firmware back to the absolute minimum to see where it produces a functioning USB port on the dev kit. I will check the `FsLowPowerFeature` in the course of that investigation.

I did indeed have `SetSysClock()` as static. What difference would that make?

mikepurvis, 2016-06-13:
We've moved on from USB to Ethernet-based comms, so this is no longer relevant to me; if others have issues, they're welcome to open a new ticket.

closed 2016-06-13 by mikepurvis