andysworkshop/stm32plus #92: Synchronous mode USART trouble

mikepurvis, 2015-02-06:
I had a successfully working Uart4 in interrupt mode that I wanting to switch away from using the interrupt. This seems like it should be easiest thing ever, but for whatever reason I'm getting parity errors triggered solidly.

Definition:

```
    struct PinPackage
    {
      enum
      {
        Port_CK=0,
        Port_TX=GPIOA_BASE,
        Port_RX=GPIOA_BASE,
        Port_CTS=0,
        Port_RTS=0,

        Pin_CK=0,
        Pin_TX=GPIO_Pin_0,
        Pin_RX=GPIO_Pin_1,
        Pin_CTS=0,
        Pin_RTS=0
      };
    };

    typedef Uart4_Custom<PinPackage> Uart;
    Uart* uart;
```

Initialization:

```
    uart = new Navsat::Uart(115200);
```

Operation:

```
    while(1)
    {
      if (uart->dataAvailable())
      {
        // Transfer from UART hardware buffer to local buffer.
        char ch = navsat_hw_.uart->receive();
        LOG_INFO("Char [%d]", ch);

        uint32_t ec;
        if (uart->hasError(ec))
        {
          LOG_INFO("Err [%d]", ec);
          continue;
        }
      }
```

Unfortunately, I'm getting the char as consistently 255, and the error as 0x1, which is USART_FLAG_PE, indicating a parity error. I'm confused how this is even possible, though, given that the default 8N1 configuration has no parity at all— what is triggering a parity error in this circumstance, and seemingly triggering it over and over, given that hasError performs the DR read necessary to clear the error flag?

Would appreciate any thoughts here.

andysworkshop, 2015-02-06:
I tried to reproduce this on an F4 discovery using your pinout (PA0/PA1) on UART4 and couldn't get it to break. This code worked as a basic synchronous echo test using a terminal program on the other end.

      typedef Uart4_Custom<UsartReceiveSyncTest> UsartType;
      UsartType *uart=new UsartType(115200);

      for(;;) {

        uint8_t b;

        while(!uart->dataAvailable());

        if((b=uart->receive())==0xff && errorProvider.hasError()) {

          // ST's error flag is in errorProvider.getCause();
          b='X';
        }

        uart->send(b);
      }

mikepurvis, 2015-02-06:
Hmm. Well, I ran out of time trying to figure this out an resumed using the interrupt.

I wonder if it's perhaps another issue stemming from using an RTOS. In my scenario, the "tight polling loop" is in reality much less tight than it appears, as it's subject to context switching with other tasks.

closed 2015-02-06 by mikepurvis