andysworkshop/stm32plus #93: I2C as slave device

punkkeks, 2015-02-06:
Hi everyone,

I have trouble to figure out how to use my stm32f030 as a slave device (in interrupt mode).  What I have so far is:

    I2C1_Custom<I2Cf030PinPackage, I2C1InterruptFeature> i2c;
Unfortunately, there are a very few (namely zero) examples for I2C (in slave mode). I found a feature ```I2CSecondSlaveFeature```, but no ```I2CSlaveFeature``` like in the f1,f4 subfolder. Is that a mistake? How I am supposed to transfer/read bytes with I2C?

andysworkshop, 2015-02-06:
I've successfully done this for `I2C2`. I'm a bit rushed at the moment so here's a quick code fragment. I'll post the whole thing this weekend.

The declaration

    typedef I2C2_Remap1<I2C2InterruptFeature> MyI2C;
    scoped_ptr<MyI2C> _i2c;

The initialisation

    MyI2C::Parameters params;

    // initialise the peripheral

    params.i2c_timing=0xB0420F13;
    params.i2c_ownAddress=I2C_ADDRESS << 1;
    params.i2c_analogFilter=false;

    _i2c.reset(new MyI2C(params));

    // subscribe to interrupts

    _i2c->I2CInterruptEventSender.insertSubscriber(
        I2CInterruptEventSourceSlot::bind(this,&CommandReader::onInterrupt));

The interrupt handler

inline void CommandReader::onInterrupt(I2CEventType eventType) {

  bool full;

  switch(eventType) {

    case I2CEventType::EVENT_ADDRESS_MATCH:
      _addressReceived=true;
      break;

    case I2CEventType::EVENT_RECEIVE:                 // data received

      // got some data

      _addressReceived=false;

      // write the byte

      _commandBuffer.write(I2C_ReceiveData(*_i2c));   // add to the circular buffer

      full=_commandBuffer.availableToWrite()==0;
      _indicators.setFull(full);                      // set/reset the full LED

      // is the buffer full? Suspend incoming if it is.

      if(full)
        _commandBuffer.suspend();

      break;

    case I2CEventType::EVENT_STOP_BIT_RECEIVED:
      if(_addressReceived)                            // no data in frame? must be a reset request
        NVIC_SystemReset();
      else
        _addressReceived=false;
      break;

    default:
      break;
  }

Hope this is enough to get you started. As you can see it's a fragment from a working program which I'll post in full in a few days.

punkkeks, 2015-02-06:
Ah, I see. There is the function ```I2C_ReceiveData``` in the periphal lib, did not know that. I hope for a similar solution like the spi or uart interface, where you have dedicated write and read methods. Maybe there is a chance to add them? ;)

closed 2015-02-11 by andysworkshop