andysworkshop/stm32plus #96: Verification for CircularBufferInputOutputStream

mikepurvis, 2015-02-19:
Looks like there aren't any examples or tests for this class. My use (writing one character at a time from an ISR, reading one character at a time from the main loop, checking `available()`) is finding that the buffer fills up, with both the read and write pointers pointing at the end of the loop, and neither reads nor writes succeed.

Investigation ongoing.

andysworkshop, 2015-02-19:
Hi Mike, can you use `memory/circular_buffer<T>` as an alternative? It's designed for your exact use case and I use it for that purpose in [this project](https://github.com/andysworkshop/awcopper).

I'll revist `CircularBufferInputOutputStream` and re-implement it so that it uses circular_buffer<T> underneath.

mikepurvis, 2015-02-19:
I looked briefly at that, but it didn't seem suitable for conveying data from an ISR, as the read and write pointers and the buffer are not volatile.

andysworkshop, 2015-02-19:
The data members don't need to be marked as `volatile` because all the accessor methods are `volatile`. It's a little unconventional I know but it has the desired effect and I've verified the assembly language output to ensure it's correct.

mikepurvis, 2015-02-19:
Interesting, okay. Will switch over when able.

closed 2015-02-21 by andysworkshop