andysworkshop/stm32plus #100: stm32f4xx_can doesn't build correctly

spiralray, 2015-03-14:
Hello.

stm32f4xx_gpio.o contains following functions correctly.

    $ arm-none-eabi-nm stm32f4xx_gpio.o
    00000000 T GPIO_DeInit
    00000000 T GPIO_Init
    00000000 T GPIO_PinAFConfig
    00000000 T GPIO_PinLockConfig
    00000000 T GPIO_ReadInputData
    00000000 T GPIO_ReadInputDataBit
    00000000 T GPIO_ReadOutputData
    00000000 T GPIO_ReadOutputDataBit
    00000000 T GPIO_ResetBits
    00000000 T GPIO_SetBits
    00000000 T GPIO_StructInit
    00000000 T GPIO_ToggleBits
    00000000 T GPIO_Write
    00000000 T GPIO_WriteBit
             U RCC_AHB1PeriphResetCmd
but, stm32f4xx_can.o doesn't contain any functions.

    $ arm-none-eabi-nm stm32f4xx_can.o

I found this issue under the following environment.

    scons mode=small mcu=f4 hse=25000000 -j4 float=hard

This problem maybe related to mcu_defines.h
Is there any solution?

andysworkshop, 2015-03-14:
Fixed on master.

This was a bug introduced in the migration to the new CMSIS version and affected all the 'optional' peripherals such as DCMI, SAI, FMC, CAN and others.

Would you be interested in stm32plus CAN support? I was thinking of designing a dev board because they're not easy to find and adding support to stm32plus.

spiralray, 2015-03-14:
Thanks for fixing the issue.

Yes, I'm interested.
CAN is becoming a popular communications protocol in robot industry. I also frequently used.

Please tell me if there's anything I can help with.

mikepurvis, 2015-03-14:
I'm interested in CAN support also.

closed 2015-05-03 by andysworkshop