andysworkshop/stm32plus #103: SPI Timing issue

punkkeks, 2015-03-26:
The following code produces the following signal trace.

    Spi1<>::Parameters params;
    params.spi_direction = SPI_Direction_1Line_Tx;
    params.spi_mode = SPI_Mode_Master;
    Spi1<> spi(params);

    spi.setNss(false);
    spi.send(initCode, 14); //14 bytes for LCD initialisation
    spi.setNss(true);

[image]

As you might see, the enable pin (5) is going high to early (round about 0.22ms)

Netherless if the spi.setNss is correct way to do it or not, this is a problem. I think we don't wait for the last byte to be sent. Maybe we want to fix that!

andysworkshop, 2015-03-26:
Yes you're right. I'll need to add a method to poll the `BSY` bit in `SPI_SR` and update the examples accordingly.

andysworkshop, 2015-04-04:
Fixed with new `waitForIdle()` method. Tested on F0/F1/F4.

closed 2015-05-03 by andysworkshop