andysworkshop/stm32plus #108: Ethernet PHY Support

danielschramm, 2015-04-14:
Hello,

Olimex changed the PHY on their STM32-E407 boards to LAN8710A, which currently is not supported by stm32plus.
I attached my quick and dirty implementation for this chip to this issue. Please review and include it into the distribution.

Thanks,

Daniel

#pragma once

namespace stm32plus {
  namespace net {

    /**
     * Class for the  LAN8710A PHY.
     */

    class LAN8710A : public PhyBase {

      public:

        /**
         * Custom registers for this PHY
         */

        enum {
        	PHY_BasicControl = 0,
        	PHY_BasicStatus = 1,
        	PHY_Identifier1 = 2,
        	PHY_Identifier2 = 3,
        	PHY_AutoNegotiationAdvertisement = 4,
        	PHY_AutoNegotiationLinkPartnerAbility = 5,
        	PHY_AutoNegotiationExpansion = 6,
        	PHY_ModeControlStatus = 17,
        	PHY_SpecialModes = 18,
        	PHY_SymbolErrorCounter = 26,
        	PHY_ControlStatusIndication = 27,
        	PHY_InterruptSource = 29,
        	PHY_InterruptMask = 30,
        	PHY_SpecialControlStatus = 31
        };

        /**
         * Interrupt bits
         */

        enum {
          INTERRUPT_ENERGYON              		= 0x80,
          INTERRUPT_AUTO_NEGOTIATION        	= 0x40,
          INTERRUPT_REMOTE_FAULT				= 0x20,
          INTERRUPT_LINK_DOWN			    	= 0x10,
          INTERRUPT_AUTO_NEGOTIATION_LP_ACK 	= 0x8,
          INTERRUPT_PARALLEL_FAULT         	 	= 0x4,
          INTERRUPT_AUTO_NEGOTIATION_PAGE_RX	= 0x2,
          INTERRUPT_ALL                       	= 0xf7
        };

        /**
         * Configuration parameters for the LAN8710A
         */

        struct Parameters : virtual PhyBase::Parameters {

          /**
           * Constructor: set defaults
           */

          Parameters() {
            phy_resetDelay      = 2;        // wait 2ms after reset is asserted
            phy_speedStatusBit  = 0x0002;   // link speed bit in status word
            phy_duplexStatusBit = 0x0004;   // full duplex bit in status word
          }
        };

        bool phyEnableInterrupts(uint8_t interruptMask) const;
        bool phyDisableInterrupts(uint8_t interruptMask) const;

        bool phyIs100M(bool& is100) const;
        bool phyIsFullDuplex(bool& isFull) const;
        bool phyClearPendingInterrupts() const;
        bool phyGetPendingInterrupts(uint16_t& interrupts) const;

        // hard reset is supported through the PhyHardReset feature class

        void phyHardReset(GpioPinRef& pin) const;

        bool initialise(Parameters& params,NetworkUtilityObjects& netutils);
        bool startup();
    };

    /**
     * Initialise the PHY class
     * @param params The parameters class
     * @param netutils The network utilities
     * @return true if it worked
     */

    inline bool LAN8710A::initialise(Parameters& params,NetworkUtilityObjects& netutils) {
      return PhyBase::initialise(params,netutils);
    }

    /**
     * Startup
     * @return true if it worked
     */

    inline bool LAN8710A::startup() {
      return PhyBase::startup();
    }

    /**
     * Check if we are linked at 100Mb/s
     * @param is100 true if we are at 100Mb/s
     * @return true if it worked
     */

    inline bool LAN8710A::phyIs100M(bool& is100) const {

      uint16_t value;

      if(!phyReadRegister(PHY_BasicStatus,value))
        return false;
      is100=false;
	  if(value & (1<<9)) { // 100BASE-T2 Half Duplex
		  is100=true;
	  }
	  if(value & (1<<10)) { // 100BASE-T2 Full Duplex
		  is100=true;
	  }
	  if(value & (1<<13)) { // 100BASE-TX Half Duplex
		  is100=true;
	  }
	  if(value & (1<<14)) { // 100BASE-TX Full Duplex
		  is100=true;
	  }
	  if(value & (1<<15)) { // 100BASE-T4
		  is100=true;
	  }
      return true;
    }

    /**
     * Check if we are linked at full duplex
     * @param is100 true if we are at 100Mb/s
     * @return true if it worked
     */

    inline bool LAN8710A::phyIsFullDuplex(bool& isFull) const {

      uint16_t value;

      if(!phyReadRegister(PHY_BasicStatus,value))
        return false;

      isFull=false;
	  if(value & (1<<10)) { // 100BASE-T2 Full Duplex
		  isFull=true;
	  }
	  if(value & (1<<12)) { // 10BASE-T Full Duplex
		  isFull=true;
	  }
	  if(value & (1<<14)) { // 100BASE-TX Full Duplex
		  isFull=true;
	  }
      return true;
    }

    /**
     * Clear pending interrupt flags. On the LAN8710A reading the MISR register
     * clears the pending interrupt flags
     * @return true if it worked
     */

    inline bool LAN8710A::phyClearPendingInterrupts() const {
      uint16_t value;
      return phyReadRegister(PHY_InterruptSource,value);
    }

    /**
     * Read the pending interrupts register to determine which ones have fired
     * @param[out] interrupts The interrupt bitmask. Can be AND'ed with INTERRUPT_ values
     * @return true if it worked
     */

    inline bool LAN8710A::phyGetPendingInterrupts(uint16_t& interrupts) const {

      if(!phyReadRegister(PHY_InterruptSource,interrupts))
        return false;

//      // pending bits are in the upper byte, shift them down so that the INTERRUPT_* masks apply
//
//      interrupts>>=8;
      return true;
    }

    /**
     * Enable interrupts to be received on the Observable interface. This function merges
     * the bits in interrupt mask into the register
     * @param interruptMask The interrupts to enable (INTERRUPT_*)
     * @return true if it worked
     */

    inline bool LAN8710A::phyEnableInterrupts(uint8_t interruptMask) const {
      return
          phySetRegisterBits(PHY_InterruptMask,interruptMask);
//		  &&
//          phySetRegisterBits(PHY_InterruptSource,interruptMask);
    }

    /**
     * Disable interrupts to be received on the Observable interface. This function masks out
     * the bits in interrupt mask into the register
     * @param interruptMask The interrupts to enable (INTERRUPT_*)
     * @return true if it worked
     */

    inline bool LAN8710A::phyDisableInterrupts(uint8_t interruptMask) const {
      return phyClearRegisterBits(PHY_InterruptMask,interruptMask);
    }

    /**
     * Hard-reset the PHY by pulling the RESET pin low for 5ms. The datasheet
     * indicates that 1us is the minimum between stable power and rising reset so
     * we're playing safe.
     * @param pin
     */

    inline void LAN8710A::phyHardReset(GpioPinRef& pin) const {
      pin.reset();
      MillisecondTimer::delay(5);
      pin.set();
    }
  }
}

mikepurvis, 2015-04-14:
I'm using this chip as well, via the DM-STF4BB "baseboard". Andy and I are presently trying to get it going via email, but I'll give your class here a shot too.

It's not clear to me what the difference is between the Microchip LAN8710A/LAN8720A and the SMSC LAN8710/LAN8720. It seems they're all compatible, though, so maybe it makes sense to just call the class LAN8710, and note in the header that it works with the variants.

mikepurvis, 2015-04-14:
Okay, your class here is working for me on both the 8710 and 8720. I can successfully run the net_dhcp and net_tcp_server examples. Thanks!

andysworkshop, 2015-04-14:
This couldn't have come at a better time, thanks Daniel! I've merged your code into the master branch with only a small efficiency change to the full duplex and 100M methods. I'm sure it should still work.

danielschramm, 2015-04-15:
Mike, you are right, this is from the press information:
"Microchip acquired SMSC in August 2012. Microchip is investing heavily in the technology to continue the proud history of leadership in integrated circuit products for computing applications."
So it's the same product.
Andy, Thanks for integrating and reviewing.

danielschramm, 2015-04-15:
Andy, I just updated to your changes and it is still working.

closed 2015-04-16 by danielschramm