andysworkshop/stm32plus #109: Parser Class

danielschramm, 2015-04-15:
Hello,
I implemented a parser class ans like to discuss the Idea if it's worth to be added to stm32plus.
Each char based interface (TCP = Telnet, USART, USB CDC, ...) uses a command prompt as user interface. All these derive from a "Communicator". 
All these communicators can be added to a parser class which compares strings to registered commands. Rest of the command string is passed to the command receiver which can reply it's answer to the communicator from which it received the command.
With this different communication channels can be used in parallel.

Daniel

danielschramm, 2015-04-15:
/*
 * parser.hpp
 *
 *  Created on: 10.08.2011
 *      Author: schramm
 */

#ifndef PARSER_HPP_
#define PARSER_HPP_

#include <stdio.h>
#include <string.h>
#include <string>
#include <list>
#include <config/event.h>
#include "singleton.hpp"

/**
 * The command line parser gets it's command lines from a Communicator, compares to
 * registered strings and replies are sent to the Communicator which provided the command
 * Communicators are also registered to the parser.
 */

class Communicator;

DECLARE_EVENT_SIGNATURE(ParserEvent, void(const std::string &, Communicator &));

class Communicator {
private:
public:
	enum CommunicationType{EEL, MNF};			//Type of communication, EEL of Manufacture (PBK) it is importent fo some commands
	enum CommunicationType CommType=MNF;
	DECLARE_EVENT_SOURCE(ParserEvent);
	void newLine(const std::string &str) {
	ParserEventEventSender.raiseEvent(str, *this);
	}
	virtual void respond(const char* str) = 0;

};

class Parser {
private:
	class OptionEntry {
	public:
		std::string command;
		fastdelegate::FastDelegate2<const std::string &, Communicator &> delegation;
	};
	typedef std::list<OptionEntry> _ParserList;
	_ParserList parserList;
	typedef _ParserList::const_iterator ParserIterator;
	typedef std::list<Communicator *> _CommunicatorList;
	_CommunicatorList communicatorList;
	typedef _CommunicatorList::const_iterator CommunicatorIterator;

	typedef wink::slot<void (int)> slot;
	wink::signal<slot> sender;

	void update (const std::string &parseString, Communicator &com) {
		for (ParserIterator i = parserList.begin(); i != parserList.end(); ++i)
		{
			uint8_t len = i->command.length();
			if(i->command.compare(0,len,parseString,0,len)==0) {
				(i->delegation)(parseString.substr(len, parseString.length()), com);
			}
		}

	}

public:
	Parser() {
		parserList.clear();
	}

	void addCommunicator(Communicator &com) {
		communicatorList.push_back(&com);
		com.ParserEventEventSender.insertSubscriber(
				ParserEventEventSourceSlot::bind(this, & Parser::update)
		);
	}
	void removeCommunicator(Communicator &com) {
		for(_CommunicatorList::iterator it=communicatorList.begin(); it!=communicatorList.end(); ++it) {
			if((*it)==&com) {
				communicatorList.erase(it);
				return;
			}
		}
	}

	void addOption(const char *str, fastdelegate::FastDelegate2<const std::string &, Communicator &> newdelegation) {
		OptionEntry newEntry;
		newEntry.command=static_cast<std::string>(str);
		newEntry.delegation=newdelegation;
		parserList.push_back(newEntry);
	}
	template< class X, class Y>
	void addOption(const char *str,  Y * obj, void (X::*func)(const std::string &p1, Communicator & ) )
	{
		addOption(str, fastdelegate::MakeDelegate( obj, func ) );
	}

	template< class X, class Y>
	void addOption(const char *str, Y * obj, void (X::*func)(const std::string &p1, Communicator & ) const )
	{
		addOption(str, MakeDelegate( obj, func ) );
	}

};

typedef Singleton<Parser> _Parser;
#define PARSER _Parser::Instance()

#endif /* PARSER_HPP_ */

andysworkshop, 2015-04-15:
Hi Daniel, I'll look into this and respond soon.

punkkeks, 2015-05-04:
Hey Daniel, would you like to put some ticks around your code (3x `) (see https://guides.github.com/features/mastering-markdown/#GitHub-flavored-markdown), or even better make a gist at http://gist.github.com

closed 2016-11-12 by andysworkshop