andysworkshop/stm32plus #110: Scheduler Class

danielschramm, 2015-04-15:
Hello, 
again for discussion as new feature. 
For my project I use a scheduler class to manage tasks.
A task (method of an object) can be registered to the scheduler to be executed every x ms.
The scheduler has two methods called regulerly, the first from systick IRQ and the second in the main program loop. During registration of a new task it can be decided to use hard realtime (called in systick IRQ for chritical timings like measurements) or as soft realtime in the main program loop for data processing and communication.
Daniel

danielschramm, 2015-04-15:
/*
 * schedulerMS.hpp
 *
 *  Created on: 13.10.2014
 *      Author: Schramm
 */

#ifndef SRC_LIB_SCHEDULERMS_HPP_
#define SRC_LIB_SCHEDULERMS_HPP_

#include <list>
#include <iterator>
#include "event/fd/FastDelegate.h"
#include "singleton.hpp"
#include "timing/MillisecondTimer.h"

extern "C" {
  extern void __attribute__ ((interrupt("IRQ"))) SysTick_Handler(void);
}

/**
 * Scheduler class is called every ms by the timer interrupt.
 * A delegate can be registered to be called every n milliseconds.
 *
 * This file overrides the millisecond IRQ from stm32plus library
 */

class SchedulerMS {
protected:
	static SchedulerMS *thisptr;
	typedef fastdelegate::FastDelegate0< void > _Delegate;
	struct SchedItem {
		uint16_t milliseconds;
		uint16_t countdown;
		bool execute;
		_Delegate delegate;
	};
	std::list<SchedItem> hardList;
	std::list<SchedItem> softList;

	void executeHard() {
		for(std::list<SchedItem>::iterator it=hardList.begin(); it!=hardList.end(); ++it) {
			if((*it).countdown>1) {
				(*it).countdown--;
			} else {
				(*it).delegate();  // execute hard directly
				(*it).countdown=(*it).milliseconds;
			}
		}
		for(std::list<SchedItem>::iterator it=softList.begin(); it!=softList.end(); ++it) {
			if((*it).countdown>1) {
				(*it).countdown--;
			} else {
				(*it).execute=true; // execute soft later
				(*it).countdown=(*it).milliseconds;
			}
		}

	}

	/**
	 * declare isr friend function
	 * allows it to access private members like thisptr
	 */
	friend void ::SysTick_Handler(void);

	void isr();

public:

	void init() {
		thisptr = this;
	}

	void update() {
		for(std::list<SchedItem>::iterator it=softList.begin(); it!=softList.end(); ++it) {
			if((*it).execute) {
				(*it).delegate();
				(*it).execute=false;
			}
		}
	}

	void addDelegateHard(const uint16_t time, _Delegate delegate) {
		SchedItem si={time, time, false, delegate};
		hardList.push_back(si);
	}
	void addDelegateSoft(const uint16_t time, _Delegate delegate) {
		SchedItem si={time, time, false, delegate};
		softList.push_back(si);
	}

	// TODO: this function is currently untested. Please test end remove this comment
	void removeDelegateSoft(const uint16_t time, _Delegate delegate) {
		for(std::list<SchedItem>::iterator it=softList.begin(); it!=softList.end(); ++it) {
			if(((*it).milliseconds == time) && ((*it).delegate==delegate)) {
				softList.erase(it);
			}
		}
	}

	// TODO: this function is currently untested. Please test end remove this comment
	void removeDelegateHard(const uint16_t time, _Delegate delegate) {
		for(std::list<SchedItem>::iterator it=hardList.begin(); it!=hardList.end(); ++it) {
			if(((*it).milliseconds == time) && ((*it).delegate==delegate)) {
				hardList.erase(it);
			}
		}
	}
};

typedef Singleton<SchedulerMS> _SchedulerMS;
#define SCHEDULER _SchedulerMS::Instance()

#endif /* SRC_LIB_SCHEDULERMS_HPP_ */

mikepurvis, 2015-04-15:
This is getting into RTOS territory. Rather than stm32plus rolling its own, I wonder if it might make sense to instead have a blessed integration scheme and examples which demonstrate how to work with one of the more popular existing choices (ChibiOS, NuttX, Contiki, maybe FreeRTOS).

andysworkshop, 2015-04-15:
Yes I'd prefer that stm32plus plays well with the 'real' RTOS systems rather than straying into their domain. If there are any parts of stm32plus that prevent or hinder the use of an RTOS then I'd certainly fix that.

danielschramm, 2015-04-16:
Hello Andy, Hello Mike,

the intention to keep stm32plus clean from OS functionality is clear, but there is already some basic timing control in STM32plus like the millisecond timer which provides "timed out", etc. I just like to start a discussion if featureas like regulary signals to objects or generate signal in "x" milliseconds might be a good idea to be added to the timing functionality giving them another name instead of tasks.

I also like the idea of Mike to show an integration sheme to one RTOS

Daniel

closed 2015-07-09 by andysworkshop