andysworkshop/stm32plus #111: Clarify buffer safety in stm32plus::net

mikepurvis, 2015-04-28:
Having some issues with my application which I believe are down to calling `net->read` and `net->write` in the manner of a Unix socket, which appears not to be safe, as the data is not copied to an internal buffer. I think I'd benefit from some clarification on the following questions:

 - Can I call `TcpClientConnection::receive` at my leisure, or only in the context of the event handler? The examples seem to go both ways on this.
 - What's the capacity of the receive behaviour? Do I have control over that? What happens when it fills up?
 - For `TcpClientConnection::send`, it looks like it's DMA code in use, which means the pointer I pass it needs to survive to some point in the future when the data is actually transferred. Is there's a blessed scheme for sending a transient buffer, or should I just roll my own?

Thanks.

andysworkshop, 2015-04-28:
> For TcpClientConnection::send, it looks like it's DMA code in use, which means the pointer I pass it needs to survive to some point in the future when the data is actually transferred. Is there's a blessed scheme for sending a transient buffer, or should I just roll my own?

DMA is used at the link layer to transmit your data in-place from your buffer but higher up in the transport layer `TcpConnection::send` waits for the ACKs to arrive back from the other side so by the time you get control back the data is guaranteed to not only have gone out to the wire but also ACK'd by the receiver. `send()` should not be called in the context of an IRQ.

> What's the capacity of the receive behaviour? Do I have control over that? What happens when it fills up?

`tcp_receiveBufferSize` in the stack's parameters object holds the maximum that can be buffered while you're away doing other things. This member automatically becomes available when you configure the TCP transport into your stack. When it fills up we advertise a zero window back to the remote end which will cause it to go into a timed back-off retry loop. This is a situation you want to avoid because it has an adverse affect on performance.

> Can I call TcpClientConnection::receive at my leisure, or only in the context of the event handler? The examples seem to go both ways on this.

It should be safe to call from both contexts but it's not the best way to receive data.

The best use of the TCP module is to use blocking `send()` - which it appears you are doing, and interrupt driven receive which I don't think you're doing. Interrupt driven receiving means you avoid the zero-window scenario and can transfer data from the network buffer to your program as fast as you get it. To get notified when data has arrived you can subscribe to the `TcpConnectionDataReady` event in the `TcpConnection` class.

mikepurvis, 2015-04-28:
Is there any advantage in using an ISR to receive if all I'm doing is moving the data out of the library's queue and into my own? If not, I'd rather just increase the library's queue size and handle the data as I am ready to do so.

Either way, it sounds like I'm using the API correctly. Thanks!

closed 2015-04-28 by mikepurvis

andysworkshop, 2015-04-29:
After refreshing my memory on this part of the code I've made an important correction. Neither `send()` nor `receive()` are IRQ-safe and should only be called from the normal context.

My original suggestion that you could use `TcpConnectionDataReadyEvent` to process incoming data is not correct. Unfortunately it's raised from an IRQ context which means that you can't then react to it by calling `receive()`. The best you could do in the event handler is to set a flag of your own to indicate that data is ready to come and fetch but then you might just as well try calling `receive()` instead of checking a flag. Not too useful I'm afraid.

mikepurvis, 2015-04-29:
I did note that both of them are potentially blocking functions, which yes, does make them unsuitable for ISR use. In any case, my application is perfectly fine with just calling them periodically from the main context.