andysworkshop/stm32plus #114: Reading a PWM signal

punkkeks, 2015-05-04:
Is it possible to read a PWM signal (more specific the duty cycle of it)? If yes, could you provide some code examples or even better, update the examples?

andysworkshop, 2015-05-04:
Yes it's possible to do that using a timer's input capture/compare facility. In [this example](https://github.com/andysworkshop/stm32plus/tree/master/examples/timer_input_capture) I use the technique to calculate a frequency but you can use it to measure a duty cycle using the feature that captures the signal on rising _and_ falling edges.

punkkeks, 2015-05-04:
Okay, I think I can work with the example. What requirements are needed for the timer? I the official example I found Tim2 (fun fact: the stm32f0 series does not have a Tim2 *weird*). And is it possible to use on timer for multiple counting? I want to read 9 PWM signals (from hall sensors). Any ideas?

closed 2016-11-12 by andysworkshop