andysworkshop/stm32plus #115: iostream/iostream.h confusion

mikepurvis, 2015-05-04:
In the included STL implementation, there's a file called `lib/include/stl/iostream`, however several files (`lib/include/stl/iterator`, `lib/include/stl/string`, others) contain an include for `<iostream.h>`.

I've been experimenting with compiling various portions of stm32plus on x86 to facilitate testing, and this is causing a headache. Any thoughts?

andysworkshop, 2015-05-09:
Generally speaking I would probably leave the stm32plus STL behind and use the one provided by the x86 environment. They're functionally the same so you should be able to use a different one without any ill effects.

I developed the FAT16/32 file system driver in this way. It was coded and debugged first on the PC using a mock block device and then transferred over to the stm32.

mikepurvis, 2015-05-10:
That was my initial instinct as well, however there was an issue with the `slist` header being required— it's an SGI extension and not in the standard implementations.

andysworkshop, 2015-05-10:
`slist` is used in stm32plus because it's lightweight compared to `list` so long as you only want to iterate forwards. On the PC that won't matter so might get away with a `typedef` from `list` to `slist` and see how that goes.

mikepurvis, 2015-05-22:
This worked great for me:

```
#if __linux__
#include "list"
namespace std
{ 
  template<typename T> using slist = list<T>;
} 
#else
#include "slist"
#endif
```

closed 2015-05-22 by mikepurvis