andysworkshop/stm32plus #116: tcpConnect is a blocking call

mikepurvis, 2015-05-07:
Any multitasking application which creates TCP connections to the outside world has to move any other periodic processing off the main thread into ISRs. This is a bummer. It would be great to have a tcpConnectAsync or similar which returns immediately and reveals the status of the connection state machine via callbacks or silent updates to the TcpClientConnection object.

mikepurvis, 2015-05-08:
A little progress to report here.

Since I'm only connecting to a device on the local network segment, I can tolerate a much shorter timeout than if I were connecting over the internet. And I am happy to manage the reconnection logic myself, so the following parameters give me a lot more control over making the connection:

```
    params_.tcp_connectRetryInterval = 100;
    params_.tcp_connectMaxRetries = 0;
```

It still blocks for 100ms waiting on the SYN/ACK, but that's a lot more acceptable for my situation than losing the main thread for seconds at a time.

Gonna close this for now, though a fully asynchronous `tcpConnect` would still be a worthwhile addition, I think.

andysworkshop, 2015-05-09:
Although you've closed it I agree that an asynchronous version would be useful and I'll add it to the list of things to do.

mikepurvis, 2015-05-12:
Hmm, looks like there are a couple other places where one can still encounter a relatively long (>1s) block when trying to `tcpConnect`. For example, when the server is no available or responding.

Errors thrown are 49/6/0, 50/3/0, and 54/4/0.

andysworkshop, 2015-05-27:
I've tested this on my F407 board and it seems to work. Please let me know if it works for you.

mikepurvis, 2015-05-29:
This looks good— I'm still in the process of trying it with my own code. Is it necessary to capture the event in the event handler? The way my program is structured, it would be more convenient to simply poll for it periodically, eg:

```
while(1) {
  if (conn && conn->getState == TcpState::ESTABLISHED) {
    // Do stuff with the active connection
  } else {
    if (time-since-last-connection-attempt) {
      if (conn) {
        delete conn;
        conn = NULL;
      }
      net->tcpConnectAsync<TcpClientConnection>("192.168.1.3", 12345, conn)
    }
  }
  // Other non-network related stuff.
}
```

I can obviously populate a volatile from the ISR, but it seems like this would more convenient.

andysworkshop, 2015-05-31:
Hi Mike, yes you can poll if you want as long as you're prepared for the possibility of missing a state change. If the server were to accept and then quickly close the connection for some reason then you might just see the closing states if your loop is doing something else at the time. Probably not an issue if you code for that eventuality.

closed 2015-07-09 by andysworkshop