andysworkshop/stm32plus #121: Unable to read from two ADC peripherals

tonybaltovski, 2015-06-12:
I am trying to use the ADC1 and ADC3 peripherals on the F4 series.  I am able to use the peripherals  individually but when I try to use the MultiDmaMultiChannel example to read from both, the readings are incorrect.  Below is a snippet of my implementation.  Any thoughts on what the issue could be?

```
#include "config/stm32plus.h"
#include "config/timing.h"
#include "config/adc.h"
#include "config/usart.h"
#include "config/string.h"

using namespace stm32plus;

class AdcMultiDmaMultiChan {

  private:
    volatile bool _ready;

  public:

    void run() {

      uint16_t dmaTransfers;

      _ready=false;

      volatile uint16_t readBuffer[13];

      Adc1DmaChannel<AdcMultiDmaMode1Feature<Adc1PeripheralTraits>,Adc1DmaChannelInterruptFeature> dma;

      Adc1<
        AdcClockPrescalerFeature<2>,
        AdcResolutionFeature<12>,
        Adc1Cycle480RegularChannelFeature<2, 3, 4, 6, 10, 11, 12, 13>,
        AdcScanModeFeature<>,
        AdcDualRegularSimultaneousDmaMode1Feature<
          Adc3<
            AdcClockPrescalerFeature<2>,
            AdcResolutionFeature<12>,
            Adc3Cycle480RegularChannelFeature<4, 5, 6, 7, 8>,
            AdcScanModeFeature<>,
            AdcContinuousModeFeature
          >,
          5
        >
      > adc;

      // half-word transfers

      dmaTransfers=13;

      dma.DmaInterruptEventSender.insertSubscriber(
          DmaInterruptEventSourceSlot::bind(this,&AdcMultiDmaMultiChan::onComplete)
      );

      dma.enableInterrupts(Adc1DmaChannelInterruptFeature::COMPLETE);

      Usart3_Remap1<> usart(57600);
      UsartPollingOutputStream outputStream(usart);

      dma.beginRead(&readBuffer,dmaTransfers);
      adc.startRegularConversion();

      for(;;) {

        for (int i = 0; i < dmaTransfers; i++)
        {
          readBuffer[0]=readBuffer[1]=readBuffer[2]=readBuffer[3]=0xAAAA;
        }

        while(!_ready);
        _ready=false;

        outputStream << "Converted values are ";
        for (int i = 0; i < dmaTransfers; i++)
        {
          outputStream  << StringUtil::Ascii(readBuffer[i]) << ", ";
        }
        outputStream << "\r\n";

        MillisecondTimer::delay(1000);
      }
    }

    void onComplete(DmaEventType det) {
      if(det==DmaEventType::EVENT_COMPLETE)
        _ready=true;
    }
};

int main() {

  Nvic::initialise();
  MillisecondTimer::initialise();

  AdcMultiDmaMultiChan adc;
  adc.run();

  return 0;
}
```

andysworkshop, 2015-06-18:
Hi Tony, sorry for taking so long to respond to this. I've had time to do a quick test myself and yes I seem to be able to cause issues when ADC3 is the slave. I'll have to refresh my knowledge on the valid combinations documented in RM0090 and I'll check the code for bugs.

tonybaltovski, 2015-06-19:
No worries, thank you Andy.  Would there be any other way to read those pins all at once?  They are not a high priority.

danielschramm, 2015-06-22:
I also have issues with 2 ADCs at the same time.
ADC1 and ADC3, both individually running on DMA.
ADC1 with Byte transfer and ADC3 with HalfWord.
This leads to wrong or missing data on one or both of the ADCs.
The code is working perfect for both channels induvidual. I will try to change the ADC3 to polling now.

tonybaltovski, 2015-06-22:
@danielschramm do you have an example of polling multiple pins on ADC3?

andysworkshop, 2015-06-22:
ADC3 cannot be paired with ADC1 as a slave in dual mode. The only valid combination in dual mode is ADC1 and ADC2.

Please refer to _Figure 51. Multi ADC block diagram_ in RM0090 and see footnote (2) at the bottom where it says "2. In the Dual ADC mode, the ADC3 slave part is not present". This is backed up by the documentation for the `MULTI[4:0]` bits in the `ADC_CCR` register.

danielschramm, 2015-06-23:
Hello Tony,
I put my code here:
<script src="https://gist.github.com/danielschramm/0038237a14d5509afca4.js"></script>

https://gist.github.com/danielschramm/0038237a14d5509afca4

Best Regards,

Daniel

mikepurvis, 2015-06-23:
Yikes— that looks like a nightmare, having to re-initialize the the peripheral every time through the loop.

Couldn't this be managed using `AdcTripleFeature`, and just reading off a dummy ADC2 pin in order to get the ADC1 and ADC3 pins you actually want?

danielschramm, 2015-06-23:
Hello Mike,
I would be glad to get a better example of a running code to learn from. 
Even better would be an explainaition why ADC1 and ADC3 kannot be run seperately with different DMA settings in Scan Mode which would be the best solution.
My code is currently doing it's job for ADC3 in parallel to ADC1 with DMA.
Daniel

andysworkshop, 2015-06-23:
>Even better would be an explainaition why ADC1 and ADC3 kannot be run seperately with different DMA settings in Scan Mode which would be the best solution.

There's no obvious reason why this should not work as long as the conversions from each ADC cannot overlap because there's only one `ADC_DR` register in single ADC mode.

tonybaltovski, 2015-07-08:
I couldn't get the `AdcTripleRegularSimultaneousDmaMode1Feature` working so I went a method similar to what @danielschramm suggested (Thanks!).

closed 2015-07-08 by tonybaltovski