andysworkshop/stm32plus #122: DHCP own IP

danielschramm, 2015-06-26:
Hello,

I'm just trying to understand the network code.
Is there a clean way to get information about the DHCP status from my Application, like which
IP, Netmask, Gateway and Nameserver were assigned?
All these dara exist as protected variables in DhcpClient.h.
Is the intended way to derive an own class from DhcpClient and add the getter functions and then use this derived client as feature in ApplicationLayer? 
Or is there another intention behind your implementation?

Thanks,

Daniel

andysworkshop, 2015-06-26:
Hi Daniel, because a DHCP transaction requires multiple steps between you and the DHCP server it is all done asynchronously in the background. When the transaction is complete you will be notified of the results. The [example code](https://github.com/andysworkshop/stm32plus/blob/master/examples/net_dhcp/net_dhcp.cpp#L156) shows how it's done.

danielschramm, 2015-06-29:
Hi Andy, thanks. I missed this part when reading network samples. 
Thanks a lot. It works great now.

closed 2015-06-29 by danielschramm