andysworkshop/stm32plus #123: redifinition of __STRICT_ANSI__

punkkeks, 2015-07-05:
Compiler output:

    arm-none-eabi-g++   -DSTM32PLUS_BUILD -DSUB_BUILD_BRIDGE=1 -Wall -ffunction-sections -fdata-sections -fno-exceptions -mthumb -gdwarf-2 -pipe -DHSE_VALUE=8000000 -mcpu=cortex-m4 -DSTM32PLUS_F407 -Os -Wextra -fno-rtti -std=gnu++0x -fno-threadsafe-statics -std=c++1y -flto -fno-use-linker-plugin -Wno-unused-parameter -I/usr/local/arm-none-eabi/include/> stm32plus-040100 -I/usr/local/arm-none-eabi/include/stm32plus-040100/stl -I/home/marcel/studium/bitbots/firmware/subboard/inc    -o CMakeFiles/bridge.dir/src/main.cpp.o -c /home/marcel/studium/bitbots/firmware/subboard/src/main.cpp
    In file included from /usr/local/arm-none-eabi/include/stm32plus-040100/stl/iterator:30:0,
                     from /usr/local/arm-none-eabi/include/stm32plus-040100/config/event.h:19,
                     from /usr/local/arm-none-eabi/include/stm32plus-040100/config/rtc.h:19,
                     from /usr/local/arm-none-eabi/include/stm32plus-040100/config/timer.h:24,
                     from /usr/local/arm-none-eabi/include/stm32plus-040100/config/timing.h:18,
                     from /home/marcel/studium/bitbots/firmware/subboard/src/main.cpp:5:
    /usr/local/arm-none-eabi/include/stm32plus-040100/stl/stl_config.h:165:0: warning: "__STRICT_ANSI__" redefined
    #define __STRICT_ANSI__
    ^
    <built-in>: note: this is the location of the previous definition
    /usr/bin/cmake -E cmake_progress_report /home/marcel/studium/bitbots/firmware/build/CMakeFiles 2

Cmakefile:

    cmake_minimum_required(VERSION 3.2)
    project(subboard C CXX ASM)
    
    set(CMAKE_PREFIX_PATH /usr/local/arm-none-eabi /usr/arm-none-eabi CACHE STRING "Library search paths")
    
    set(MCU_FAMILY f407)
    set(STM32PLUS_CONFIGURATION small-${MCU_FAMILY}-8000000)
    find_package(stm32plus REQUIRED)
    include_directories(${STM32PLUS_INCLUDE_DIRS} inc)
    
    set(CMAKE_SHARED_LIBRARY_LINK_C_FLAGS "")
    set(CMAKE_SHARED_LIBRARY_LINK_CXX_FLAGS "")
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -flto -fno-use-linker-plugin -Wno-unused-parameter")
    
    set(SOURCE
        src/main.cpp
        src/subboard.cpp
        src/odroid.cpp
        src/ledring.cpp
        src/dogl.cpp
        src/touchscreen.cpp
        system/libraryHacks.cpp
        system/${MCU_FAMILY}/system.c
        system/${MCU_FAMILY}/startup.s
    )
    
    # practically we build the same executable twice, but in fact they will differ because of the SUB_BUILD_TYPE defined
    # below. We strongly rely on the effect, that the compiler will erase all none needed code paths
    # When we compile it with either -Os or -O2/3 this will good. If not, we will try -flto and if that breaks too
    # we have to think of an other solution :) But for now, this will work!
    foreach(TRGT bridge periph)
        add_executable(${TRGT} ${SOURCE})
        target_link_libraries(${TRGT} ${STM32PLUS_LIBS})
        set_target_properties(${TRGT} PROPERTIES LINK_FLAGS "-T${PROJECT_SOURCE_DIR}/system/${MCU_FAMILY}/Linker.ld")
        add_bin_target(${TRGT})
    
    
        add_custom_target(flash_${TRGT}
            COMMAND bash -c "echo -e reset init\\\\nflash write_image erase ${CMAKE_BINARY_DIR}/${TRGT}.bin 0x8000000 bin\\\\nreset |nc localhost 4444"
            DEPENDS ${TRGT}.bin
            VERBATIM
        )
    
    endforeach()
    
    set_target_properties(bridge PROPERTIES COMPILE_DEFINITIONS "SUB_BUILD_BRIDGE=1;$<$<CONFIG:Debug>:SUB_DEBUG=1>")
    set_target_properties(periph PROPERTIES COMPILE_DEFINITIONS "SUB_BUILD_PERIPH=1;$<$<CONFIG:Debug>:SUB_DEBUG=1>")

There seems to be nothing weird at all. I can't figure out why
1. there is a redifnintion
2. you define __STRICT_ANSI__

andysworkshop, 2015-07-09:
I think you might be getting `__STRICT_ANSI__` as a builtin from your C++14 option (`-std=c++1y`). Some of the code paths in the STL are guarded by checks for `__STRICT_ANSI__` hence my definition of it. I should be protecting this definition with an `#ifndef` guard block and will do that now...

punkkeks, 2015-07-14:
Please review my comment in your commit and then close this issue, if you want.

closed 2015-07-15 by andysworkshop