andysworkshop/stm32plus #124: Update stdperiph for f0 chips

punkkeks, 2015-07-05:
The library uses version 1.3, currently in use is 1.5.
Here's the changelog

V1.5.0 / 05-December-2015
=========================

Main Changes:
-------------------
* Add support of the STM32F070xB, STM32F070x6 and STM32F030xC  devices.
* stm32f0xx_flash.c/.h: update for STM32F070xB, STM32F070x6 and STM32F030xC devices

V1.4.0 / 24-July-2014
=====================

Main Changes:
------------------
* Add support of the STM32F091 devices.
* stm32f0xx_dma.c/.h: update for STM32F091 devices
  * Add the support of the DMA2 channels.
  * Add new function "void DMA_RemapConfig(DMA_TypeDef* DMAy, uint32_t DMAx_CHy_RemapRequest);"
* stm32f0xx_flash.c/.h: update for STM32F091 devices
  * Add definition for new option bytes on the STM32F091 devices.
* stm32f0xx_gpio.c/.h: update for STM32F091 devices
  * Update the GPIO_AF definition comments with the new supported alternate functions
* stm32f0xx_rcc.c/.h: update for STM32F091 devices
  * Add USART3CLK_Frequency fields in the RCC_ClocksTypeDef structure.
  * Add definition for USART3 clock sources
  * Add USART4, USART5, USART6, USART7 and USART8 clock and reset configuration.
  * Add DMA2 clock configuration.
* stm32f0xx_syscfg.c/.h: update for STM32F091 devices
  * Add new function "void SYSCFG_IRDAEnvSelection(uint32_t SYSCFG_IRDAEnv);"
  * Add new function "uint32_t SYSCFG_GetPendingIT(uint32_t ITSourceLine);"
* stm32f0xx_usart.c/.h:  update for STM32F091 devices
  * Add support of the USART3,USART4, USART5, USART6, USART7 and USART8.
* stm32f0xx_pwr.c 
  * Fix PWR_EnterSTANDBYMode() to not clear Wakeup flag (WUF), this flag need to be cleared at application level before to call this function 
  * PWR_EnterSleepMode(): update usage of __WFE() in low power entry function. If there is a pending event, calling __WFE() instruction, the CortexM0 core will not enter to sleep mode. The solution is to make the call below: the first __WFE() is always ignored and clears the event if one was already pending, the second is always applied
 
    __SEV()
    __WFE()
    __WFE()

andysworkshop, 2015-07-09:
Implementing this on branch `f0-stdperiph-1.5`

andysworkshop, 2015-08-01:
Fully merged to master

closed 2015-08-01 by andysworkshop