andysworkshop/stm32plus #125: stm32plus with STM32CubeF4

molnard, 2015-07-10:
Hi,

I am using STM32F427VG with custom board (PCB). I have configured an eclipse project using the new 1.7.0 STM32CubeF4 library. 
I have successfully compiled stm32plus library. It looks like stm32plus also contains an older cube library. Is it possible to make it work with the new library? 
Or do I have to choose between new cube library or stm32plus?

Thank you in advance,
Dave

andysworkshop, 2015-07-11:
stm32plus packages the standard peripheral library for the f0, f1 and f4 and I try to keep those up to date with changes by ST. It does not package the newer 'cube' framework except for a few files from the F4 'HAL' that were necessary to get the USB support working.

I notice that ST have upgraded the F4 standard peripheral library from the 1.4.0 that is currently used in stm32plus to 1.5.1. I will start working shortly on upgrading this.

molnard, 2015-07-13:
Thank you for your answer!

closed 2015-07-13 by molnard