andysworkshop/stm32plus #129: HSI instead of HSE

punkkeks, 2015-08-08:
If I understand the scons help correctly, I have to use `HSE=8000000` if I would like to use the internal clock, instead of an external one, right? But what if I have an external clock that has 8MHz?
I would prefer a simple switch (e.g. HSI) to use the internal clock. 

For my current project it is not possible (due to my fault of the pcb design) to use an external clock for my stm32f407, but I'd like to experiment with the board, because everything else just works. Is there a chance to do implement this feature?

andysworkshop, 2015-08-09:
I agree. It is desirable to have a clear distinction between HSI and HSE in the build.

andysworkshop, 2015-08-11:
^^ that commit should take care of it. `hsi` and `hse` are both acceptable `scons` build options. Note that the small but important different in the name of the built library. For example `libstm32plus-small-f051-8000000e.a` is a build with `hse` and `libstm32plus-small-f051-8000000i.a` is a build with `hsi`.

punkkeks, 2015-08-12:
@mikepurvis should adapt the cmakelists as well, but it looks good. I can give you feedback on saturday.

punkkeks, 2015-08-15:
Seems like it doesn't work

    examples/adc_analog_watchdog/system/f407_168_8/System.c: In function 'SystemCoreClockUpdate':
    examples/adc_analog_watchdog/system/f407_168_8/System.c:299:25: error: 'HSE_VALUE' undeclared (first use in this function)
           SystemCoreClock = HSE_VALUE;
                             ^
    examples/adc_analog_watchdog/system/f407_168_8/System.c:299:25: note: each undeclared identifier is reported only once for each function it appears in
    examples/adc_analog_watchdog/system/f407_168_8/System.c: In function 'SetSysClock':
    examples/adc_analog_watchdog/system/f407_168_8/System.c:149:21: error: 'HSE_VALUE' undeclared (first use in this function)
     #define PLL_M      (HSE_VALUE/1000000)
                         ^
    examples/adc_analog_watchdog/system/f407_168_8/System.c:384:20: note: in expansion of macro 'PLL_M'
         RCC->PLLCFGR = PLL_M | (PLL_N << 6) | (((PLL_P >> 1) -1) << 16) |

punkkeks, 2015-08-22:
Works for me now. Thanks.

closed 2015-08-22 by punkkeks