andysworkshop/stm32plus #131: SystemCoreClock not set properly

punkkeks, 2015-08-08:
I might found a bug in the Framework (or a not very well documented issue).

I have currently no external clock for my stm32f407, so according to the `system_stm32f4xx.c` line 264 `SystemCoreClock` should contain the HSI_VALUE `16MHz`, but in my case it contained `168.000.000`. So my timings are all shitty and nothing worked at first glance. After setting the variable via gdb to `16.000.000` everything just worked fine and so on. My guess is: When there is no external clock, a softwareinterrupt will be fired (according to Datasheet page 22). I think you have to call `SystemCoreClockUpdate` in that interrupt to set the `SystemCoreClock` is set up properly. 

What do you think. Where is the failure and how to fix that? Maybe it has something to do with #129 . Hope to hear from you soon :)

punkkeks, 2015-08-08:
I just tried to call `SystemCoreClockUpdate` as the first function in my main() function a voilà... It just worked. Funny and I'm happy. But netherless that problem exists

andysworkshop, 2015-08-09:
That's quite strange behaviour. I've used the HSI with the F429 and numerous F0 versions and have never seen that happen.

punkkeks, 2015-08-10:
What do you need to reproduce the problem or what do you want me to do, to help you? I'd really like to see this "bug" fixed

andysworkshop, 2015-08-11:
Ideally a minimal sample project for the F4 discovery (F407) would be great.

punkkeks, 2015-08-22:
I fixed it by using the [clock configuration tool][0] and replacing the system.c with it.
Thanks for your help :)

[0]: http://www.st.com/web/en/catalog/tools/FM147/CL1794/SC961/SS1533/PF257927?s_searchtype=keyword#

closed 2015-08-22 by punkkeks