andysworkshop/stm32plus #135: blink example building problem in Eclipse.

burakenez, 2015-09-08:
I build stm32plus and stm32plus-example-buttons project without any error in Eclipse according to Andy's article. However, stm32plus-example-blink gave me error. Although I choose "Build Configurations/Set Active/Debug_f407_168_8" for blink project and build it, I was getting error, because of "f1md". I solved the problem by deleting one line code in "makefile" that is inside of "~/stm32plus/examples/blink/Debug_f407_168_8/".

-include system/f1md/subdir.mk

I am not very familiar with makefiles and I don't know the reason why this makefile is generated with this line of code, but I think I solved the problem by deleting it since f1md should not be there at the first place and I decide to delete it because "button" project doesn't have it.

Maybe I am the reason for that at the first place. What could be the problem which caused such an error?

andysworkshop, 2015-09-12:
I'll fix this. The problem is due to the resource exclusions not being set up for the Eclipse example projects after the merge of the "f1md" pull request.

andysworkshop, 2015-09-13:
All eclipse example projects should now be fixed.

closed 2015-09-13 by andysworkshop