andysworkshop/stm32plus #138: Getting E_TOO_BIG from my network port?

mikepurvis, 2015-10-27:
Another curious little issue from my most recent project— seeing this periodically, always immediately preceding a hard crash of the MCU. My network device is trading small, fixed-size UDP packets with a PC, and also opening a single TCP connection to a server on the PC, over which bidirectional messages are exchanged which vary in size from 50-200 bytes.

I can supply a packet capture of this happening, but I just wanted to get a quick sanity check what would cause this, and what my possible resolutions are— is it something I can catch and handle on the MCU, or should I experiment with limiting my MTU from Linux?

andysworkshop, 2015-10-27:
I don't think a packet capture is going to show the cause. `E_TOO_BIG` is thrown _prior_ to transmission of a frame if that frame would exceed the configured MTU size, the default for which is 1518 bytes.

Are you able to set breakpoints in `MacBase.cpp` on the two lines that return the `E_TOO_BIG` error code? The associated `NetBuffer` object will show how much data you're trying to send.

mikepurvis, 2015-10-27:
Yeah, should be able manage that— does raise the alarming possibility that this is something deeper, like a corrupted data structure. FWIW, all the writes (both TCP and UDP) happen from the main thread, and it's all synchronous, so there shouldn't be much scope for error.

Also odd that it hangs the micro when this happens, given that it _appears_ the error has been successfully handled and recovered from. Anyhow, we'll see more when I can take a look with the debugger.

andysworkshop, 2015-10-28:
> ...raise the alarming possibility that this is something deeper, like a corrupted data structure

This was my thought. Catching it in the debugger will reveal all but I suspect you're going to see memory corruption causing a wild value to be in the length field and that corruption goes on to cause the hard fault shortly after.

mikepurvis, 2015-11-10:
Ended up switching my TCP socket to also use UDP— it was a better fit for the use case anyway, so I didn't end up sorting out whether this was an issue with my application or the library, unfortunately.

closed 2015-11-10 by mikepurvis