andysworkshop/stm32plus #139: CAN support

andysworkshop, 2015-11-14:
CAN is a major peripheral on the STM32 family and we don't yet support it in stm32plus.

andysworkshop, 2015-11-18:
@Elektroman2 I've reviewed your pull request and have commited it to a new branch `can-support`. Can you verify that it still works for you?

Also, are you planning to commit a simple CAN bus example?

Elektroman2, 2015-11-18:
I found it, I will send everything tomorrow. ( I messed up things with git :( )

andysworkshop, 2015-11-28:
thanks to @spiralray we've now got F4 support and I'm about ready to merge `can-support` to `master` unless anyone's got any objections...

andysworkshop, 2015-11-30:
`can-support` is now merged to `master`. F0 support will complete this feature but I'll close this issue now and track F0 support under a new issue.

closed 2015-11-30 by andysworkshop