andysworkshop/stm32plus #149: Pluggable delay method?

mikepurvis, 2016-01-12:
I [raised this briefly here](https://github.com/andysworkshop/stm32plus/issues/19#issuecomment-165983571), but I'd like to file a separate issue— when integrating with an RTOS, it's critical that stm32plus's `delay` method be a yield rather than just a busy-wait. A few thoughts for how to implement this:

 - Supply a function that allows registering an alternative delay implementation.
 - Supply a function that allows supplying an alternative MillisecondTimer implementation.
 - Have `delay` call a weakref'd function that the user can supply an alternative implementation of.

Thoughts?

punkkeks, 2016-01-30:
Maybe use the [`weak`-attribute](https://gcc.gnu.org/onlinedocs/gcc-4.6.4/gcc/Function-Attributes.html#index-g_t_0040code_007bweak_007d-attribute-2555) of gcc?

mikepurvis, 2016-02-06:
I'm interested in this. Would you entertain a PR for a weak-attribute based solution?

It's unclear to me if weak linking can work for a static class method like MillisecondTimer, or if it would need to pass through to a separate free function. Either way, the `inline` is going to be gone, whether effectively or explicitly.

andysworkshop, 2016-02-09:
`weak` could work. The only problem with `weak` is that when multiple implementations are `weak` then it's undefined who wins. That shouldn't be an issue here though.

andysworkshop, 2016-03-24:
closed by #169

closed 2016-03-24 by andysworkshop