andysworkshop/stm32plus #150: ARP lookup always fails until an initial packet is received

nnarain01, 2016-01-12:
When attempting to send a UDP packet to a PC the ARP lookup always fails until an initial packet is received from the PC.

The error callback gives:

    Error Provider: 49, Code: 6 (ARP, Timeout)
    Error Provider: 50, Code 3  (IP, ARP lookup failed)

Network Stack Setup

	typedef stm32plus::net::PhysicalLayer<stm32plus::net::LAN8710A> PhysicalLayer;
	typedef stm32plus::net::DatalinkLayer<
		PhysicalLayer,
		stm32plus::net::DefaultRmiiInterface,
		stm32plus::net::Mac> DatalinkLayer;
	typedef stm32plus::net::NetworkLayer<
		DatalinkLayer,
		stm32plus::net::DefaultIp,
		stm32plus::net::Arp
	> NetworkLayer;
	typedef stm32plus::net::TransportLayer<
		NetworkLayer,
		stm32plus::net::Udp,
		stm32plus::net::Icmp
	> TransportLayer;
	typedef stm32plus::net::ApplicationLayer<
		TransportLayer,
		stm32plus::net::StaticIpClient,
		stm32plus::net::Ping
	> ApplicationLayer;
	typedef stm32plus::net::NetworkStack<ApplicationLayer> NetworkStack;

Network Parameters

		params_.base_rtc = rtc_;
		params_.staticip_address        = MCU_IP;
		params_.staticip_subnetMask     = MCU_NETMASK;
		params_.staticip_defaultGateway = GATEWAY_IP;
		params_.ping_timeout = PING_TIMEOUT;

I figure that when a packet is received its IP is added to the ARP cache, and that's why things start working.

However I'm wondering what the possibilities could be that would cause the lookup to fail.

andysworkshop, 2016-01-13:
> I figure that when a packet is received its IP is added to the ARP cache, and that's why things start working

That's correct. The reason for an ARP lookup timeout is that the ARP probe sent out by stm32plus was not received by the PC or the response from the PC was not received by stm32plus. If the PC in question is not on the same network segment as stm32plus then that could be the reason. Either way, can you run Wireshark and set the filter to "arp" and watch the traffic? You should be able to see if the probe reaches the PC and if the PC responds. Unfortunately you won't be able to see if a reponse makes it to stm32plus.

nnarain01, 2016-01-13:
Thank you. After looking at the arp packet I realized I needed to set the gateway after the address of my PC.

closed 2016-01-13 by nnarain01

andysworkshop, 2016-01-13:
This should not be necessary. The code does a match using your PC's IP address against the subnet mask received from either DHCP lookup or static configuration. If your PC's IP address matches the subnet then it's local and an ARP probe is issued to get the MAC address. Otherwise the ARP probe is issued for the gateway's MAC address so that the frame can be sent to a different subnet.

Can you verify that your PC's IP address is covered by the subnet mask?