andysworkshop/stm32plus #156: [Discussion] Restructure stm32plus

punkkeks, 2016-01-30:
Hey everyone,

I'd like to reconstructure the library a bit. I really love to work with it, but sometimes it annoys me a lot.
Some examples

 - Inlcude paths
   -  have to write `#include "config/spi.h"`. I would prefer something like `#include <stm32plus/spi.h>`
 - Using a newer std library
   - the current SGCS Library is quiet old (nearly 20 years now)
   - no support of modern c++ and no >= c++11 features
   - I am playing aroud with llvm's libc++. I managed to build everything with libc++ and it works good and the compiled binarys are equal in size or even smaller (some files are bigger, but not a big deal)
 - Static compilation
   - I don't like, that you use one big static compiled library, instead of (maybe one) generic shared one. Of course you have to create a static binary in order to make a bin/hex dump, but the library itself could be dynamic I think.
 - Determine the cpu and features at compile time of the program, not the library
   - I don't like the fact, that I have to rebuild the library three times, because I develop software for three types of cpu atm. I think this can be a difficult thing.
 - Switch the build system
   - I really don't like scons. I know this is a personal issue. But maybe there's a chance to switch to cmake ;)
 - Allow the library to build with clang
   - This is my current site. It works, but then suddly scons swallows the CCFLAGS and everything breaks. Another argument for switching to CMake

I know this is a lot and I think, when we discussed about this, it would be good to create an issue for each section and implement it on a different branch, because it would brake things. But I think this would improve the usage of the library a lot, espacially the c++11 features. You can get rid of some legacy code (scoped_array, scoped_ptr => std::unique_ptr), and some algorithmns, like none_of, all_of, etc...

andysworkshop, 2016-01-31:
There's some great ideas in there and I'm open to at least some of them :) Let's start with upgrading the standard library which I agree is past it's sell-by date. If you have everything built with a newer version then please submit a PR and I'll merge it to a new branch for testing.

mikepurvis, 2016-01-31:
Ah, didn't realise there was anyone else using CMake with stm32plus. Apart from your preference that it not be pre-compiled, how do you find the current config module scheme?

punkkeks, 2016-02-01:
@mikepurvis I like the scheme. It's easy and it finds (nearly every time) the newest version. Sometimes it does not find it, if you have multiple versions installed, but I don't know whether it's a fault of cmake or mine. I just fix that issue by deleting the old version.

@andysworkshop I'm working on it. Hang on, it's not that easy to do it right :D

baadc0de, 2016-03-16:
+1 for an upgraded std library. I don't particularly mind scons, but sure, cmake is fine as well. Allowing clang is a good idea to me too.

cszawisza, 2016-11-28:
Hi!
what is the status of "transformation"? :)

punkkeks, 2016-11-28:
To be honest, I didn't manage to implement any of this points, except building it against libc++ . The problem there was the huge config file, which you have to modify in order to get a good result in terms of usability and size for libc++ (e.g. disable thread features, don't use I/O, no underlying system, using libc or muscle.... That's a quiet big file there).

I wouldn't mind if somebody says, that he/she wants to implement any of this features, but currently I'm not capable of doing any of these (work, Christmas, holidays, ...) sorry for that.

yaqwsx, 2017-06-29:
Hi, is there any progress?

punkkeks, 2017-07-01:
I'm afraid not. I guess you're welcome to do any work ;) I would help, wherever I can, but tbh i didnt work with this library over 4 months by now, because auf other projects.

Tal-seven, 2018-12-30:
@hellow554 : Sorry for revisiting this old thread. But i really like some of the points you've mentioned at the top. Especially "Determine the cpu and features at compile time of the program, not the library" . This has been a sticking point for me as well. Do you have any idea on how this could be implemented or what type of an approach would be appropriate. I'm willing to try it out. Any suggestions would help. Cheers