andysworkshop/stm32plus #159: C++ syntax question

dholth, 2016-02-01:
At https://github.com/andysworkshop/stm32plus/blob/master/lib/include/dma/features/f0/DacDmaWriterFeature.h#L20 it says "If the setup of the initialisation structure is not to your needs then it can be customised before the transfers are started by casting this class to |DMA_InitTypeDef|". What's the syntax for doing this?

Second question, Is Eclipse fairly useless at autocomplete for template classes or is mine just configured incorrectly?

Thanks.

andysworkshop, 2016-02-06:
Is this what you're looking for (F0 example)?

      Dac1<DacChannel112BitRightAlignmentFeature> dac;
      Dac1UpdateDmaChannel<DacDmaWriterFeature<DacChannel112BitRightAlignmentFeature>> dma;

      static_cast<DMA_InitTypeDef&>(dma).DMA_Mode=DMA_Mode_Circular;

> Is Eclipse fairly useless at autocomplete for template classes or is mine just configured incorrectly

It's not great. It'll find the template names for me but it won't list the parameters.

dholth, 2016-02-06:
I did get dma circular to dac to play an .au file or a sine wave. I'll probably submit an example.

closed 2016-02-09 by dholth