andysworkshop/stm32plus #166: Build error. cstdlib: no such file or directory

WheresMyMind, 2016-03-06:
Hi.
I try to build the library, but get an error:
```
../stm32plus.h:147:19: fatal error: cstdlib: No such file or directory
#include <cstdlib>
```

I installed gcc-arm-none-eabi and scons.
What could be the problem?

[image]

WheresMyMind, 2016-03-06:
Solved.
Need add folder with cstdlib in include path.

closed 2016-03-06 by WheresMyMind

andysworkshop, 2016-03-06:
You shouldn't need to add anything - gcc knows where its standard headers are. There's something not right with your compiler installation.

WheresMyMind, 2016-03-06:
I'm install  gcc  as here:
[](https://travis-ci.org/andysworkshop/stm32plus/jobs/110735759)

` sudo add-apt-repository ppa:terry.guo/gcc-arm-embedded -y`
`sudo apt-get update`
`sudo apt-get install gcc-arm-none-eabi`

mikepurvis, 2016-03-06:
As explained on the PPA page, the version numbers are such that Ubuntu Trusty's version of gcc-arm-none-eabi takes priority over the one from Terry Guo. You can resolve this one of two ways: First, specify the exact version to install:

```
sudo apt-get install gcc-arm-none-eabi=4.9.3.2015q1-0trusty13
```

Or use my PPA, which has the versions backported from Xenial, which do have superseding version numbers, and then also install the C++ stdlib:

```
sudo add-apt-repository ppa:mikepurvis/stm32-staging -y
sudo apt-get update
sudo apt-get install gcc-arm-none-eabi libstdc++-arm-none-eabi
```

WheresMyMind, 2016-03-07:
@mikepurvis, thanks. It works.

mikepurvis, 2016-03-07:
@WheresMyMind Which method did you use?

@andysworkshop Might be worth a note about this in the README. Not sure how many other Ubuntu users you have, but this is a non-obvious pitfall to navigate.

WheresMyMind, 2016-03-08:
I'm install this version:

`sudo apt-get install gcc-arm-none-eabi=4.9.3.2015q3-1vivid1`

andysworkshop, 2016-03-08:
@mikepurvis I can add a note in the README. Is this a situation unique to Ubuntu users or is it for all Linux distros?

mikepurvis, 2016-03-08:
It's specific to Ubuntu 14.04+.

punkkeks, 2016-03-08:
Don't say that. I'm not 100% sure, but on Fedora you can install the
compiler without the stdlib for arm (which makes... sense, at least a
little bit).
So, just mention it in the Readme.

danielkucera, 2018-06-13:
On Ubuntu 17.10 I needed to install:
`sudo apt-get install libstdc++-arm-none-eabi-newlib`