andysworkshop/stm32plus #170: Initializing network stack resets

mikeodr, 2016-03-31:
We're running into an issue where our networking stack is successfully initializing without a network cable/other endpoint being powered on, but upon attempting to send a [ping](https://github.com/andysworkshop/stm32plus/blob/b3de5c15d85c561f880f29faf9e60cfee1a0d52a/lib/include/net/application/ping/Ping.h#L127) we get a reset.

Has anyone else run into this issue? This is reproducible either with:
* A cable not being plugged in
* The end point not being powered up.

Our use case is that our board is what switches power to the other endpoint. We are trying to use ping to detect it is ready for network communications for us to tx/rx with it. Unfortunately this is leading to a bit of a reset loop of our board. 

Any thoughts or workarounds appreciated.

mikeodr, 2016-03-31:
We're using `net/physical/lan8710a/LAN8710A.h`

mikepurvis, 2016-03-31:
I wonder if `phyIsLinked()` could help here? I was unable to use it for this purpose on a previous project because the connection was through a switch (which was always powered on), but it should do the trick if the connection is direct.

Maybe there's some more graceful way for `stm32plus::net` to recover from this scenario, though?

mikeodr, 2016-03-31:
Apologies for the issue.
This appeared to be our own watchdog.

For other peoples benefit the default ARP timeout is 5seconds with 5 retires in a wait loop. 

Loop [here](https://github.com/andysworkshop/stm32plus/blob/master/lib/include/net/network/arp/ArpCache.h#L283)

And timeout settings [here](https://github.com/andysworkshop/stm32plus/blob/master/lib/include/net/network/arp/Arp.h#L52)

closed 2016-03-31 by mikeodr

andysworkshop, 2016-04-02:
Let me know if I can be any help here, i.e. please log a feature request if there's something else that you think the stack can provide.