andysworkshop/stm32plus #173: Missing way to set TIM_CtrlPWMOutputs from timer feature classes.

mikepurvis, 2016-06-13:
Calling this function is part of the [deprecated `initCompare` scheme](https://github.com/andysworkshop/stm32plus/blob/290d523b846fee89cafdda1a15b39b2492ac013a/lib/include/timer/features/TimerChannel1Feature.h#L88), but there's no way to do it using the feature classes.

This breaks initializing TIM1/8 as h-bridge motor controllers. Here's what you end up having to do:

```
template<class Pins>
struct timer_features<Timer1, Pins>
{
  typedef Timer1InternalClockFeature InternalClockFeature;
  typedef Timer1CustomGpioFeature<
      TIM1_CH1_OUT<Pins>,
      TIM1_CH1N<Pins>,
      TIM1_CH2_OUT<Pins>,
      TIM1_CH2N<Pins> > GpioDriveFeature;

  static void initDrive()
  {
    // Gah!
    TIM_CtrlPWMOutputs(tim, ENABLE);
  }
};

template<template <typename...> class DriveTimer, class TimerPins>
class DriveHardware
{
  typedef timer_features<DriveTimer, TimerPins> DriveTimerFeatures;

  template< template <typename...> class TimerChannelFeature >
  using DriveChannelFeature = TimerChannelFeature<
    TimerChannelOCModeFeature<TIM_OCMode_PWM1>,
    TimerChannelOCPolarityFeature<TIM_OCPolarity_High>,
    TimerChannelOCNPolarityFeature<TIM_OCPolarity_High>,
    TimerChannelOCPulseFeature<0> >;

  typedef DriveTimer<
    typename DriveTimerFeatures::InternalClockFeature,
    typename DriveTimerFeatures::GpioDriveFeature,
    DriveChannelFeature<TimerChannel1Feature>,
    DriveChannelFeature<TimerChannel2Feature>
  > DriveTimerT;

  DriveHardware(....)
  {
    [ ... ]
    // Eeep!
    DriveTimerFeatures::initDrive();
  }
}
```

With a feature class for enabling PWM output, the static function and oddball constructor call would be able to go away.

andysworkshop, 2016-07-26:
I'll add this as part of a more general feature class to control the break functionality offered by some of the timers. I guess it's a safety feature that the `MOE` bit is initialized low after reset so you have to know about it even if you're not using the break feature that it was designed to support.

andysworkshop, 2016-07-31:
Mike - see the `timer_pwm_break` demo. The call you want is `timer.enableMainOutput();` that you get from the `UnlockedTimerBreakFeature` template.