andysworkshop/stm32plus #174: STM32F7 support?

0rel, 2016-06-19:
Your library looks really promising and well written, and I'd like try it out with a small project of mine (actually I'm trying to rebuild your Vivaz Display/R61523 test with my DIY STM32F7 board). Is there a way to add support for the newer STM32F7x MCUs, by adapting the STM32F4x code? - Since I'm a STM32 newbie, I think it's too difficult for me to port this myself right now...

punkkeks, 2016-06-19:
See #148 
But maybe something changed in the meantime

0rel, 2016-06-19:
Ok thanks! Too bad there's this huge gap between the Standard Peripheral Library and HAL... I'll try it out anyway on the STM32F4 Discovery board, and see if I can port the R61523 driver portion over to the STM32F7 (will be shared again).

closed 2016-06-19 by andysworkshop