andysworkshop/stm32plus #177: compile error on arm-none-eabi-gcc 6.1.1

idreamerhx, 2016-07-11:
arm-none-eabi-gcc (Arch Repository) 6.1.1 20160526

scons mode=debug mcu=f1md hse=8000000

scons: Reading SConscript files ...
stm32plus build version is 040004
scons: done reading SConscript files.
scons: Building targets ...
arm-none-eabi-g++ -o examples/adc_analog_watchdog/build/debug-f1md-8000000e/adc_analog_watchdog.o -c -Wextra -pedantic-errors -fno-rtti -std=gnu++0x -fno-threadsafe-statics -Wall -Werror -ffunction-sections -fdata-sections -fno-exceptions -mthumb -gdwarf-2 -pipe -DHSE_VALUE=8000000 -mcpu=cortex-m3 -DSTM32PLUS_F1_MD -O0 -g3 -Ilib/include -Ilib/include/stl -Ilib -Iexamples/adc_analog_watchdog examples/adc_analog_watchdog/adc_analog_watchdog.cpp
In file included from lib/include/stl/vector:36:0,
                 from lib/include/config/string.h:20,
                 from examples/adc_analog_watchdog/adc_analog_watchdog.cpp:12:
lib/include/stl/stl_bvector.h: In member function 'std::vector<bool, _Alloc>::iterator std::vector<bool, _Alloc>::erase(std::vector<bool, _Alloc>::iterator)':
lib/include/stl/stl_bvector.h:809:5: error: this 'if' clause does not guard... [-Werror=misleading-indentation]
     if (__position + 1 != end())
     ^~
lib/include/stl/stl_bvector.h:811:7: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'if'
       --_M_finish;
       ^~
cc1plus: all warnings being treated as errors
scons: *** [examples/adc_analog_watchdog/build/debug-f1md-8000000e/adc_analog_watchdog.o] Error 1
scons: building terminated because of errors.

idreamerhx, 2016-07-11:
if (__position + 1 != end()) 
 { copy....}

andysworkshop, 2016-07-16:
Thanks for the headsup. When gcc 6 becomes a release at the ARM launchpad site then I'll address compatibility issues.

closed 2016-07-16 by andysworkshop

martin31821, 2016-08-15:
I'm experiencing similar problems when trying to compile using GCC6.1 on archlinux.
Problems I ran into and their fix (for now):
- Misleading indentation as mentioned in the first post (add -Wno-misleading-indentation to SConstruct)
- negative shift as mentioned in the second comment (add -Wno-shift-negative-value to SConstruct)
- Error using the CRC library:
  - Invalid conversion from int to stm32plus::Endian (something with -fpermissive, haven't found a solution yet.)

trorornmn, 2016-08-27:
I have same issue with gcc6.1.1 on archlinux...
Is there any progress on this?
I was able to solve first two problems by adding warning options as @martin31821 said, but how should I deal with the final problem 'Invalid conversion from int to stm32plus::Endian'?
or should I use previous version of GCC?

martin31821, 2016-08-27:
For the last issue, the only help I found was to remove -Werror and -Wpedantic, but this is no solution to the problem, it just tells the compiler to treat these warnings not as errors.

If you get the library itself to build, there are several other build failures in the examples (more specifically timers and pwm)

andysworkshop, 2016-08-29:
The currently supported gcc release is [documented here](http://andybrown.me.uk/forum/index.php/topic,126.0.html?PHPSESSID=t097ichj26a5jsud5g18u9lea0)

Scytie, 2017-01-26:
According to 

> "Thanks for the headsup. When gcc 6 becomes a release at the ARM launchpad site then I'll address compatibility issues."

6.2.1 is now released on [Launchpad](https://launchpad.net/gcc-arm-embedded/)

andysworkshop, 2017-01-27:
On launchpad [this page](https://launchpad.net/gcc-arm-embedded/6.0/6-2016-q4-major) has a blank downloads section for me. I did find a [better looking page](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads) at the ARM website and will download it from there to take a look.