andysworkshop/stm32plus #194: Compile Issue

denizea, 2017-03-01:
Hi there, 
I have been checking the the repo and really like it. Now I would like to give a try. However, I am having a problem with compiling Scons script with current output. Do you have any recommendation about what I am missing?

Thank you

$ scons mode=debug mcu=f042 hsi=8000000 install
scons: Reading SConscript files ...
stm32plus build version is 040006
scons: done reading SConscript files.
scons: Building targets ...
arm-none-eabi-g++ -o examples/adc_single/build/debug-f042-8000000i/adc_single.o -c -Wextra -pedantic-errors -fno-rtti -std=gnu++0x -fno-threadsafe-statics -Wall -Werror -ffunction-sections -fdata-sections -fno-exceptions -mthumb -gdwarf-2 -pipe -DHSI_VALUE=8000000 -mcpu=cortex-m0 -DSTM32PLUS_F0_42 -O0 -g3 -Ilib/include -Ilib/include/stl -Ilib -Iexamples/adc_single examples/adc_single/adc_single.cpp
In file included from lib/include/stl/vector:36:0,
                 from lib/include/config/string.h:20,
                 from examples/adc_single/adc_single.cpp:11:
lib/include/stl/stl_bvector.h: In member function 'std::vector<bool, _Alloc>::iterator std::vector<bool, _Alloc>::erase(std::vector<bool, _Alloc>::iterator)':
lib/include/stl/stl_bvector.h:809:5: error: this 'if' clause does not guard... [-Werror=misleading-indentation]
     if (__position + 1 != end())
     ^~
lib/include/stl/stl_bvector.h:811:7: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'if'
       --_M_finish;
       ^~
cc1plus.exe: all warnings being treated as errors
scons: *** [examples/adc_single/build/debug-f042-8000000i/adc_single.o] Error 1
scons: building terminated because of errors.

denizea, 2017-03-01:
OK, apparently, latest version of compiler has some issues with this error. 

Downgrading to https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q1-update/+download/gcc-arm-none-eabi-5_3-2016q1-20160330-win32.exe helped me. 

I hope this helps for someone who is having the same issue.

closed 2017-03-01 by denizea

punkkeks, 2017-03-01:
Just remove the -Werror in the scons makefile and it should work