andysworkshop/stm32plus #196: Build errors on recent GCC release(s)

Krakonos, 2017-03-23:
I tried building with the latest arm embedded toolchain, which contains GCC 6.3.1. However, I found a few issues, some of them I wasn't able to solve:

1) Bad indentation in lib/include/stl/stl_bvector.h:808
2) Shifting signed values in lib/src/display/graphic/PicoJpeg.cpp:281
3) constexpr casting in b/lib/include/timer/features/f0/Timer1GpioFeature.h

Issues 1 and 2 are easy to fix, I've posted a PR #195 which should get rid of these in a civilized matter. However, I can't seem to find a good solution for issue 3, as it seems the C++ standard does not allow to reinterpret_cast a constexpr value. Anyway, I have no idea what are these files generated from.

andysworkshop, 2017-04-09:
merged #195. I'll fix the reinterpret_cast problem.

andysworkshop, 2017-04-22:
All fixed now and travis updated to build using gcc 6.

closed 2017-04-22 by andysworkshop