andysworkshop/stm32plus #202: bug r61523.cpp

mado1362, 2017-10-17:
hi i found a problem r61523.cpp line 209.
_gl->drawBitmap();

andysworkshop, 2017-10-17:
Please include a full description of the issue and the problem that it causes.

closed 2017-10-17 by andysworkshop

mado1362, 2017-10-18:
Hi. I used the library STM32PLUS directly in the project. But then I found
out I had to make it with CYGWIN.
Is it okay? Would it be possible to send me the steps to do it more easily.
I am a beginner in cpp. Thank you for answering my message. I need your
help. I hope that I will Accept your student's title. I'm proud of you.

mado1362, 2017-10-21:
If my question is irrelevant, please tell me.

Regard
rodsarabi

2017-10-17 20:52 GMT-07:00 ابوالفضل رودسرابی <a.rodsarabe@gmail.com>:

phildimond, 2017-10-21:
Andy did an excellent zero-to-hero article on getting his systems working. I suspect it's still valid. 

http://andybrown.me.uk/2015/03/22/stm32dev-windows/

If you follow that you will end up with a functioning installation. That's how I set mine up.

mado1362, 2017-10-21:
I can use eclipse neone instead of eclipse kepler ???

Regard
Mado

mado1362, 2017-10-21:
I install kepler new.but plug-in gnu arm on eclipse kepler not installed

2017-10-21 0:51 GMT-07:00 ابوالفضل رودسرابی <a.rodsarabe@gmail.com>:

> I can use eclipse neone instead of eclipse kepler ???
>
> Regard
> Mado
>

mado1362, 2017-10-21:
please:answer.!!

Regard
Mado

2017-10-21 1:33 GMT-07:00 ابوالفضل رودسرابی <a.rodsarabe@gmail.com>:

>
>
>
>