andysworkshop/stm32plus #206: Lubuntu - STM32 Build Error

hariudkmr, 2017-12-05:
hudkmr@hudkmr-LifeBook-V1020:~/c++workspace/github/stm32plus$ scons mode=fast mcu=f1hd hse=8000000 install
scons: Reading SConscript files ...
stm32plus build version is 040100
scons: done reading SConscript files.
scons: Building targets ...
arm-none-eabi-g++ -o examples/adc_analog_watchdog/build/fast-f1hd-8000000e/adc_analog_watchdog.o -c -Wextra -pedantic-errors -fno-rtti -std=gnu++0x -fno-threadsafe-statics -Wall -Werror -ffunction-sections -fdata-sections -fno-exceptions -mthumb -gdwarf-2 -pipe -DHSE_VALUE=8000000 -mcpu=cortex-m3 -DSTM32PLUS_F1_HD -O3 -Ilib/include -Ilib/include/stl -Ilib -Iexamples/adc_analog_watchdog examples/adc_analog_watchdog/adc_analog_watchdog.cpp
In file included from lib/include/config/timer.h:173:0,
                 from lib/include/config/timing.h:18,
                 from examples/adc_analog_watchdog/adc_analog_watchdog.cpp:8:
lib/include/timer/features/f1/Timer1GpioFeature.h: In constructor 'stm32plus::TIM1_ETR<TRemapLevel>::TIM1_ETR()':
lib/include/timer/features/f1/Timer1GpioFeature.h:28:85: internal compiler error: unexpected expression '(uint32_t)(1073741824)' of kind cast_expr
       static constexpr const uint32_t ports[4]={ GPIOA_BASE,GPIOA_BASE,0,GPIOE_BASE };
                                                                                     ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
scons: *** [examples/adc_analog_watchdog/build/fast-f1hd-8000000e/adc_analog_watchdog.o] Error 1
scons: building terminated because of errors.
hudkmr@hudkmr-LifeBook-V1020:~/c++workspace/github/stm32plus$

andysworkshop, 2017-12-09:
Hi, which version of gcc are you using?

hariudkmr, 2017-12-11:
Hi, I had switched to Linux mint(due to crash) and was able to compile the library.

Initially I had faced cstlib error and was able to overcome by installing libstdc++-arm Library

closed 2017-12-11 by hariudkmr