andysworkshop/stm32plus #208: Is STM32 L4 Supported (NUCLEO-L432KC)

WillGreen, 2017-12-19:
The [NUCLEO-L432KC](http://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-eval-tools/stm32-mcu-eval-tools/stm32-mcu-nucleo/nucleo-l432kc.html) is by far the most powerful board ST produce in the small Nucleo-32 format. This format is ideal for using with breadboards and in compact designs. However, it uses the low power STM32 L4 [STM32L432KC](http://www.st.com/en/microcontrollers/stm32l432kc.html) rather than an STM32 F4. 

Will stm32plus work with this board? I understand things like clock speeds and memory configuration will need setting up, but is the library fundamentally compatible, or is the L4 too different? 

Any information would be useful before I decide whether to order a board.

hellow554, 2017-12-19:
This is related to #174 . Maybe you can answer the questions in the issue?

WillGreen, 2017-12-19:
The difference is that the STM32 L4 is based on the same Cortex M4F as the STM32 F4. The F7 is a completely different MCU design: Cortex-M7F. So, I think these are quite different questions.

hellow554, 2017-12-19:
Maybe my answer was a little bit unclear, sorry. 
Is there a standard peripheral library for the L4?
If yes, I see nothing that would be against implementing the L4, else wise it would be difficult because if the current implementation design of this library

WillGreen, 2017-12-19:
@hellow554 now I understand! Apologies for any confusion.
Alas it seems the L4 is too new for the standard peripheral libraries. So I guess I'm out of luck.

The options would appear to be:
a) Going with L4 and not using stm32plus
b) Building my own small F4 board

Thanks for your help.

andysworkshop, 2017-12-20:
I've just tacked on a NUCLEO-L496ZG-P to my latest Digikey order. It's a bit larger than the one you referred to but the core's the same. I can't promise anything soon but we'll see how it goes.

WillGreen, 2017-12-21:
Thanks. I look forward to hearing how you get on with the L496Z.