andysworkshop/stm32plus #212: Issue with indexing when starting a new stm32plus eclipse project from examples.

andrimarjonsson, 2018-01-24:
Hi,

I know this is most likely not the right place for posting this issue but as I'm unable to register an account on the forums I figure this to be the only place left to find people experienced in dealing with stm32plus using eclipse.

Some time in the past I used the blink example as a skeleton for a small test program I was doing for a stm32f4 dev board I had and it worked beautifully. Basically copied the blink example from the examples directory to a new location and imported that project into a fresh workspace, cleaned up the include directories and that was it

Then two days ago I decided to have a look at another stm32 dev board I had,  this time with an f103 onboard and a nice small LCD. Made a few code changes to one of the lcd examples and got it working. At that point I felt it was time to do the same procedure and copy out the example and import it into a workspace, which I did, except now eclipse flat out refuses to index the damn thing. Keeps spitting out "Invalid template argument" for the Fsmc16BitAccessMode and LGDP453x_Portrait_262K classes. It however builds fine, just the IDE features of eclipse don't work since the C++/C indexer chokes on those two classes.

At first I thought it was some simple config difference but the more I examine this the less sense it makes to be honest. 

What I've tried so far:
- Deleted the index multiple times to force a re-index.
- Read countless posts about fixing C++11 indexing issues in eclipse
    - Most of them are from +3 years ago and involve adding __GXX_EXPERIMENTAL_CXX0X__ to the symbols. 
    - Since I'm using Neon.3 I suspected this would not work but tried anyway as I was desperate at that point.
- Comparing the two .cproject files using diff, the are identical apart from basic f103 and f4 differences.

When I saw the .cproject files were essentially identical I decided, since the project was all still in one cpp file, to switch main.cpp files between the projects. Might as well try and see since the working project was not dealing with those classes mentioned above. After that I forced a reindex and it just worked.

I'm completely at a loss how two basically identical .cproject files, loaded by the same eclipse version, pointing to the same gcc toolchain,indexing the same file can achieve two different results. Is there some cache I'm running into that I don't know about, please note when I say "force re-index" what I do is effectively re-import the project into a fresh workspace.

Eclipse version: Neon.3 Release (4.6.3), Build id: 20170711-2049
stm32plus version: 040100

I've attached the two .cproject files if that helps.
[workingcproject.txt](https://github.com/andysworkshop/stm32plus/files/1661716/workingcproject.txt)
[nonworkingcproject.txt](https://github.com/andysworkshop/stm32plus/files/1661717/nonworkingcproject.txt)

If I left any important piece of information out, please don't hesitate to ask.

Again, sorry for posting this here but as I said before I'm unable to register to the forum and this is driving me absolutely nuts.

andysworkshop, 2018-02-12:
Unfortunately it seems nobody else has seen this behaviour.

closed 2018-02-12 by andysworkshop

andrimarjonsson, 2018-02-18:
It would seem so, sigh. There goes my 50th attempt at using Eclipse for anything more complex than a HelloWorld example.