andysworkshop/stm32plus #225: scons 3 - examples do not compile

ucJoyfullComp, 2020-07-13:
due to changes in Scons 3 and up the object properties func_xxx are changed to __xxx__, all SConscript in the examples need to change func_code to __code__ (see line 18 in SConscript of examples/adc_single as an example).

jcw, 2020-10-27:
On macOS, SCons 4.0.1 (via homebrew), I get this output:

```
stm32plus (master) $ scons mode=small mcu=f407 hse=8000000 float=hard
scons: Reading SConscript files ...
stm32plus build version is 040100
AttributeError: 'function' object has no attribute 'func_code':
  File "/Users/jcw/code/git/stm32plus/SConstruct", line 267:
    SConscript("examples/SConscript",exports=["mode","mcu","osc","osc_type","osc_def","env","systemprefix","INSTALLDIR","INSTALLDIR_PREFIX","VERSION"])
  File "/usr/local/Cellar/scons/4.0.1_1/libexec/lib/python3.9/site-packages/SCons/Script/SConscript.py", line 661:
    return method(*args, **kw)
  File "/usr/local/Cellar/scons/4.0.1_1/libexec/lib/python3.9/site-packages/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/usr/local/Cellar/scons/4.0.1_1/libexec/lib/python3.9/site-packages/SCons/Script/SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/jcw/code/git/stm32plus/examples/SConscript", line 17:
    SConscript(example+"/SConscript",
  File "/usr/local/Cellar/scons/4.0.1_1/libexec/lib/python3.9/site-packages/SCons/Script/SConscript.py", line 661:
    return method(*args, **kw)
  File "/usr/local/Cellar/scons/4.0.1_1/libexec/lib/python3.9/site-packages/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/usr/local/Cellar/scons/4.0.1_1/libexec/lib/python3.9/site-packages/SCons/Script/SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/jcw/code/git/stm32plus/examples/adc_analog_watchdog/SConscript", line 18:
    sconscriptFile=(lambda x:x).func_code.co_filename
```

Not sure this is related, but I haven't used SCons before (I use PlatformIO, which wraps it under the hood), not sure where the problem is and how to deal with it.

Cheers,
-jcw

jcw, 2020-10-27:
To follow up - the problem is due to examples files using a Python 2 style iso v3. This batch edit fixes it:

`sed -i '' s/func_code/__code__/ examples/*/SConscript`