andysworkshop/stm32plus #226: ERROR - build stm32plus inside the terminal

FiiHaa, 2020-07-14:
Hi all,

I am trying to use stm32plus library, but I am failing in building the library inside the terminal. I follow steps  
 as is described in:
https://andybrown.me.uk/2015/03/22/stm32dev-windows/

I am running Ubuntu 

$ sudo git clone https://github.com/andysworkshop/stm32plus.git
Cloning into 'stm32plus'...
remote: Enumerating objects: 17, done.
remote: Counting objects: 100% (17/17), done.
remote: Compressing objects: 100% (16/16), done.
remote: Total 21028 (delta 2), reused 5 (delta 1), pack-reused 21011
Receiving objects: 100% (21028/21028), 21.43 MiB | 1.68 MiB/s, done.
Resolving deltas: 100% (14160/14160), done.
$ :~/src$ cd stm32plus/
$ :~/src/stm32plus$ scons mode=debug mcu=f407 hsi=8000000 -j12 install INSTALLDIR=~/install/stm32plus
scons: Reading SConscript files ...
stm32plus build version is 040100
AttributeError: 'function' object has no attribute 'func_code':
  File "/home/fiihaa/src/stm32plus/SConstruct", line 267:
    SConscript("examples/SConscript",exports=["mode","mcu","osc","osc_type","osc_def","env","systemprefix","INSTALLDIR","INSTALLDIR_PREFIX","VERSION"])
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 660:
    return method(*args, **kw)
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/fiihaa/src/stm32plus/examples/SConscript", line 17:
    SConscript(example+"/SConscript",
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 660:
    return method(*args, **kw)
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/fiihaa/src/stm32plus/examples/adc_analog_watchdog/SConscript", line 18:
    sconscriptFile=(lambda x:x).func_code.co_filename

kmani314, 2020-07-30:
I solved this by downgrading to scons v3.0. It seems like some features have been deprecated in newer releases, so downgrading to a version released around when the install guide was written fixed it for me.

jcw, 2020-10-27:
See #225 - a batch edit (of some 80+ files) solves this, then it works with SCons 4.0.1.